MartyMacGyver/ESP32-Digital-RGB-LED-Drivers #4: Using this code renders I2C unstable on ESP32/arduino?

marcmerlin, 2017-03-14:
Thanks for the code, it works great, except that as soon as I use it on my project, talking to I2C causes a crash:
0x400d65ee: i2cRead at /home/merlin/Arduino/hardware/espressif/esp32/cores/esp32/esp32-hal-i2c.c line 379
0x400d3e3e: TwoWire::requestFrom(unsigned char, unsigned int, bool) at /home/merlin/Arduino/hardware/espressif/esp32/libraries/Wire/src/Wire.cpp line 143
0x400d3e98: TwoWire::requestFrom(int, int) at /home/merlin/Arduino/hardware/espressif/esp32/libraries/Wire/src/Wire.cpp line 143
0x400d44e6: IoTuz::i2cexp_read() at /home/merlin/Arduino/libraries/IoTuz/IoTuz.cpp line 130

If I disable your code, then my I2C read which has been working for over a month, works again.

Are you using the same part of the chip, or is there some kind of incompatibility between using RMT and I2C?

MartyMacGyver, 2017-03-16:
Do you have any sample code? What pins are you assigning for I2C and for the ws2812 output? I won't be able to test this til after this weekend, but more info will allow me to see if I can reproduce the problem. If I2C and RMT.dont play well together that'll be disappointing... I have enough trouble with I2C on the ESP32 as it is (in terms of timing and reliability).

marcmerlin, 2017-03-16:
Yeah, I've been trying to narrow it down to something simple that causes a crash on top of your demo1 code, but so far no luck. And yet, the more I move the LED code inside my code (that does a lot more), I get all kinds of Wire/I2C crashes. 
I need to work on this more and will get back to you.
3 examples of different crashes below:

0x40083a21: xQueueGenericReceive at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./queue.c line 1998
0x400d638a: i2cRead at /home/merlin/Arduino/hardware/espressif/esp32/cores/esp32/esp32-hal-i2c.c line 379
0x400d3bda: TwoWire::requestFrom(unsigned char, unsigned int, bool) at /home/merlin/Arduino/hardware/espressif/esp32/libraries/Wire/src/Wire.cpp line 143
0x400d3c34: TwoWire::requestFrom(int, int) at /home/merlin/Arduino/hardware/espressif/esp32/libraries/Wire/src/Wire.cpp line 143
0x400d4282: IoTuz::i2cexp_read() at /home/merlin/Arduino/libraries/IoTuz/IoTuz.cpp line 130
0x400d42a4: IoTuz::read_encoder_button() at /home/merlin/Arduino/libraries/IoTuz/IoTuz.cpp line 160
0x400d1856: setup() at /home/merlin/prg/arduino/ESP32/IoTuz/examples/fulldemo/fulldemo.ino line 448
0x400e2482: loopTask(void*) at /home/merlin/Arduino/hardware/espressif/esp32/cores/esp32/main.cpp line 11 (discriminator 1)

0x400d61c8: i2cWrite at /home/merlin/Arduino/hardware/espressif/esp32/cores/esp32/esp32-hal-i2c.c line 379
0x400d3c0c: TwoWire::endTransmission(unsigned char) at /home/merlin/Arduino/hardware/espressif/esp32/libraries/Wire/src/Wire.cpp line 143
0x400d3c40: TwoWire::endTransmission() at /home/merlin/Arduino/hardware/espressif/esp32/libraries/Wire/src/Wire.cpp line 143
0x400d3fa5: IoTuz::pcf8574_write_(unsigned char) at /home/merlin/Arduino/libraries/IoTuz/IoTuz.cpp line 104
0x400d3fc3: IoTuz::i2cexp_clear_bits(unsigned char) at /home/merlin/Arduino/libraries/IoTuz/IoTuz.cpp line 115
0x400d4006: IoTuz::get_touch() at /home/merlin/Arduino/libraries/IoTuz/IoTuz.cpp line 200
0x400d16ee: get_selection() at /home/merlin/prg/arduino/ESP32/IoTuz/examples/fulldemo/fulldemo.ino line 448
0x400d393b: loop() at /home/merlin/prg/arduino/ESP32/IoTuz/examples/fulldemo/fulldemo.ino line 634
0x400e247d: loopTask(void*) at /home/merlin/Arduino/hardware/espressif/esp32/cores/esp32/main.cpp line 11 (discriminator 1)

0x400d65b3: i2cSetFrequency at /home/merlin/Arduino/hardware/espressif/esp32/cores/esp32/esp32-hal-i2c.c line 379
0x400d3b51: TwoWire::begin(int, int, unsigned int) at /home/merlin/Arduino/hardware/espressif/esp32/libraries/Wire/src/Wire.cpp line 143
0x400d5a0c: Adafruit_ADXL345_Unified::begin() at /home/merlin/Arduino/libraries/Adafruit_ADXL345/Adafruit_ADXL345_U.cpp line 291
0x400d41ab: IoTuz::begin() at /home/merlin/Arduino/libraries/IoTuz/IoTuz.cpp line 325
0x400d17ab: setup() at /home/merlin/prg/arduino/ESP32/IoTuz/examples/fulldemo/fulldemo.ino line 448
0x400e249e: loopTask(void*) at /home/merlin/Arduino/hardware/espressif/esp32/cores/esp32/main.cpp line 11 (discriminator 1)

marcmerlin, 2017-03-16:
@MartyMacGyver so I think I'm hitting a compiler bug or something, adding your code causes crashes in I2C later reliably. Those crashes go away if I disable your code, _or_ if I comment out 1 lines of simple code:
    pixel_x = constrain(pixel_x + move_x, 2, 318);
#if 0
    pixel_y = constrain(pixel_y + move_y, 2, 238);
#endif
that does not get run at all before the crash happens way way earlier in setup()

marcmerlin, 2017-03-16:
Will be discussing this on https://gitter.im/espressif/arduino-esp32 and will report back

MartyMacGyver, 2017-03-16:
I'm not sure what constrain() is but subtle problems like this sometimes point to code bugs like uninitialized pointers (in your code, my code, or perhaps the base Arduino-ESP32 code...). A compiler bug is possible but IMHO less likely. This code seems pretty straightforward.

marcmerlin, 2017-03-16:
Thanks. Good point on the uninitialized pointer suggestion.
Is there a way to compile ESP32 code with the equivalent of eletric fence (http://elinux.org/Electric_Fence) or something more fancy like valgrind?
Having the code crash in the place where the bug actually is sure would make things a lot easier :)

MartyMacGyver, 2017-03-16:
I've never tried that on this platform.

marcmerlin, 2017-03-17:
Ok, so I did more debugging and basically my code crashes soon _after_ I run     ws2812_setColors(NUMPIXELS, pixels);
(with NUMPIXELS=2) 
If move that statement around, it crashes after the first call.
If I did the function to return early (before   copyToRmtBlock_half() )
then the code never crashes
I just don't understand what that code does very well, so it's harder to debug from there.
If you have suggestion let me know, if not I'll look further and try to understand what the code does.

MartyMacGyver, 2017-03-17:
I can't test this on hardware til next week, but on what schedule are you reading your I2C device? Polled or interrupt driven? The RMT scheme uses interrupts and I wonder if that is a factor here.

marcmerlin, 2017-03-17:
I'm talking to the ESP32 Wire library, stuff like this:
    Wire.beginTransmission(I2C_EXPANDER);
    Wire.write(dt);
    error = Wire.endTransmission();
I'm not sure how it works behind the scenes, @me-no-dev might know, but I'm going to assume that it's interrupt driven.

MartyMacGyver, 2017-03-17:
It doesn't sound like you're doing interrupt processing there (e.g. a sensor that signals data readiness by triggering an interrupt pin), just normal polling and writing. But it might be happening at a lower level in the HAL.

Let us know what gpio pins you have things wired to. It might matter.

marcmerlin, 2017-03-17:
Right, I'm not having ISRs on top to talk to I2C.
Mmmh, actually my only ISR reds simple IO pins for a rotary encoder:
https://github.com/marcmerlin/IoTuz/blob/master/IoTuz.cpp#L70
My pin mapping is here:
https://github.com/marcmerlin/IoTuz/blob/master/IoTuz.h#L50

marcmerlin, 2017-03-17:
Ok, did a bit more narrowing down and basically it got down to 3 lines of code in this function
If I comment out the RMT lines and the SemaphoreTake, my code continues to work.
If I uncomment the semaphore take (without RMT being enabled), code hangs there.
If I uncomment the RMT lines and the semaphore take, then the LEDs change color and the code crashes soon after that. Mmmh, now and I'm not sure why, it seems to hang after the semaphore take too.
As one last test, I tried removing all the semaphore lines, but the code still hangs soon after the RMT commands are run.
Hope this helps.

void ws2812_setColors(uint16_t length, rgbVal *array) {
 (...)
  ws2812_sem = xSemaphoreCreateBinary();
  //RMT.conf_ch[RMTCHANNEL].conf1.mem_rd_rst = 1;
  //RMT.conf_ch[RMTCHANNEL].conf1.tx_start = 1;
  //xSemaphoreTake(ws2812_sem, portMAX_DELAY);
  vSemaphoreDelete(ws2812_sem);
  ws2812_sem = NULL;

marcmerlin, 2017-03-17:
I'm just confused as to why even the single xSemaphoreTake(ws2812_sem, portMAX_DELAY); line causes the code to hang when the RMT lines are also commented out.
And still, I can re-enable all that code, comment out a single line of code in a totally different place (code that does _not_ get run) and then everything works:
    //pixel_y = constrain(pixel_y + move_y, 2, 238);

MartyMacGyver, 2017-03-17:
Sounds like a race condition...

marcmerlin, 2017-03-17:
between which components? (assuming you have a guess)

marcmerlin, 2017-03-18:
I did another round of tests, because this is driving me crazy :)
At this point I think you're right and there seems to be a problem with how the semaphore works. Sadly I don't quite understand the semaphore code.
Please look at those tests below, hopefully they will put you in the right direction

I've done 3 tests to figure out what's going on, and maybe with the results, you can venture a guess better than me.
My code does this:
    Serial.println("Before colors");
    ws2812_setColors(NUMPIXELS, pixels);
    Serial.println("After colors");

Your code either works, hangs on the semaphore, or succeeds but causes I2C in my code to crash later, or even Serial.println to hang half way through a print. Here are the 3 combinations. Does that give you a better hint of what's going on?

---------------- Test #1
  ws2812_sem = xSemaphoreCreateBinary();
  RMT.conf_ch[RMTCHANNEL].conf1.mem_rd_rst = 1;
  RMT.conf_ch[RMTCHANNEL].conf1.tx_start = 1;
//  xSemaphoreTake(ws2812_sem, portMAX_DELAY);

Before colors
Af <- hang (note that it sets the LED, and then causes a hang in the middle of Serial.print later

---------------- Test #2
  RMT.conf_ch[RMTCHANNEL].conf1.mem_rd_rst = 1;
  RMT.conf_ch[RMTCHANNEL].conf1.tx_start = 1;
  xSemaphoreTake(ws2812_sem, portMAX_DELAY);

Before colors
After colors
LEDs turned on, setting up Accelerometer next
/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./queue.c:1444 (xQueueGenericReceive)- assert failed!
0x40083a21: xQueueGenericReceive at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./queue.c line 1998
0x400d68bb: i2cSetFrequency at /home/merlin/Arduino/hardware/espressif/esp32/cores/esp32/esp32-hal-i2c.c line 379
0x400d4969: TwoWire::begin(int, int, unsigned int) at /home/merlin/Arduino/hardware/espressif/esp32/libraries/Wire/src/Wire.cpp line 143

---------------- Test #3
//  RMT.conf_ch[RMTCHANNEL].conf1.mem_rd_rst = 1;
//  RMT.conf_ch[RMTCHANNEL].conf1.tx_start = 1;
  xSemaphoreTake(ws2812_sem, portMAX_DELAY);

Before colors
<= hang likely on xSemaphoreTake, never returns

projectgus, 2017-03-22:
This code may be the problem (or part of it):
https://github.com/MartyMacGyver/ESP32-Digital-RGB-LED-Drivers/blob/master/arduino-esp32/demo1/ws2812.cpp#L284

I'm guessing the intention is that ws2812_buffer is not freed until after the RMT interrupt has fully copied it into the RMT peripheral memory?

As implemented, it has at least one race condition causing a use-after-free. This may may lead to the kind of crashes you're seeing. The race is - if the RMT interrupt happens after vSemaphoreDelete, but before ws2812_sem is nulled out, then xQueueGiveFromISR in the ISR will access freed memory.

Rather than create and delete the semaphore each time, I'd suggest creating it once for the life of the driver, and use `xSemaphoreGive(ws2812_sem, 0)` at the point where you currently re-create it.

Although you may also want to think about whether you can structure this code without the semaphore at all - using a queue of RMT items or something similar.

MartyMacGyver, 2017-03-22:
Interesting - this is where I'm leaning on @FoxxTexx's original code (https://github.com/FozzTexx/ws2812-demo) so it's a latent defect. I will look at how to fix this. At the very least getting rid of the race condition will be the first step.

When you say a queue of RMT items, do you have an example or reference for how that might work?

me-no-dev, 2017-03-22:
[This](https://github.com/espressif/esp-idf/blob/master/components/mdns/mdns.c#L186) is an example where raw udp packet is pushed to a queue so it can be polled later. And [this](https://github.com/espressif/esp-idf/blob/master/components/mdns/mdns.c#L302) is where that happens. I hope it helps :)

marcmerlin, 2017-03-22:
@projectgus a huge thank you for the analysis and finding the issue
@me-no-dev thanks for the pointers on better ways to do this.
For what it's worth, after I upgraded to the latest esp-idf, my code that I had tweaked to randomly not trigger the bug, started failing instantly.
@MartyMacGyver looking forward to trying your new code and thanks again for writing that library to start with.

MartyMacGyver, 2017-03-22:
Thank @FozzTexx... he wrote it, I just took it and improved upon it. (That said, whatever I do to make it work right will need to be fed back to his as well, for completeness.) First I need to be able to reproduce the issue on my own setup, so I can at least see if I can make the problem go away with said changes. Will work on it some tonight.

\* And there is no small irony that your code is what led me to find this code which in turn you ended up using. I thought I recognized your name here... :-D

marcmerlin, 2017-03-23:
@MartyMacGyver yes I just realized 2 days ago that you were the same person on the adafruit issue saying that my patch wasn't so good (which indeed is true) :)
On reproducing, how about you send proposed code my way? I can reproduce at will here.

MartyMacGyver, 2017-03-23:
@marcmerlin Yeah, I didn't mean to hate on your code - it's just that bit-banging becomes bang-head-on-wall-ing with the ESP32. If it wasn't for @FozzTexx's original code to use the RMT peripheral I might've given up.

(As it is, I'm dealing with fundamental I2C issues on this platform that are really vexing, but that's another matter - out of curiosity, have you had any issues with I2C devices dropping offline, glitching/sending corrupt data, or otherwise acting up versus other devices like the Teensy?)

marcmerlin, 2017-03-23:
@MartyMacGyver no worries, no offence taken, you an me-no-dev were both right that it was a dead end  route, just the quickest one I could take for now, and still the one I'm using until I can use your code again.
For I2C, my board has an I/O expander I'm currently writing a lot to, to do PWM on my LCD's backlight (behind the I2C port expander) and it seems solid so far.

MartyMacGyver, 2017-03-23:
@marcmerlin While I can't reproduce the errant behavior on my setup (probably because I'm not doing I2C in an ISR), I did make the changes suggested by @projectgus (but I'm still using semaphores for now). If you try to take the semaphore before the `RMT.conf_ch[RMTCHANNEL].conf1.tx_start = 1;` part it doesn't work at all, so I left it as it is except I'm only creating the semaphore once and never delete it (though at some point I intend to make this a class and put the sem delete and the buffer free in the destructor).

Give branch **bugfix_i2c_crash** a try and let me know if it works for you - if it does, I'll integrate that change as well as a small LED timing fix (better timing compliance to the spec for the WS2812B, though I suspect it won't have any visible effect whatsoever).

marcmerlin, 2017-03-23:
Thanks for the attempt. Sadly I get the same result:. 
Guru Meditation Error of type LoadProhibited occurred on core  1. Exception was unhandled.
Register dump:
PC      : 0x400d88a6  PS      : 0x00060230  A0      : 0x800d4f45  A1      : 0x3ffc7a50  
A2      : 0x3ffc11d0  A3      : 0x00000190  A4      : 0x00000064  A5      : 0x000000c8  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x800d889b  A9      : 0xffffc000  
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x3ffc4134  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0xff000000  SAR     : 0x00000001  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

Decoding 9 results
0x400d88a6: i2cSetFrequency at /home/merlin/Arduino/hardware/espressif/esp32/cores/esp32/esp32-hal-i2c.c line 379
0x400d88a6: i2cSetFrequency at /home/merlin/Arduino/hardware/espressif/esp32/cores/esp32/esp32-hal-i2c.c line 379
0x400d4f45: TwoWire::begin(int, int, unsigned int) at /home/merlin/Arduino/hardware/espressif/esp32/libraries/Wire/src/Wire.cpp line 136
0x400d633c: Adafruit_ADXL345_Unified::begin() at /home/merlin/Arduino/libraries/Adafruit_ADXL345/Adafruit_ADXL345_U.cpp line 291
0x400d4833: IoTuz::begin() at /home/merlin/Arduino/libraries/IoTuz/IoTuz.cpp line 350
0x400d1eb7: setup() at /home/merlin/prg/arduino/ESP32/IoTuz/examples/fulldemo/fulldemo.ino line 445
0x400e5b12: loopTask(void*) at /home/merlin/Arduino/hardware/espressif/esp32/cores/esp32/main.cpp line 17

MartyMacGyver, 2017-03-23:
I wonder if your code will run without anything attached? I'm thinking I might try it that way just to see what it does. If I could reproduce the effect it'd save a lot of time iterating on it.

marcmerlin, 2017-03-23:
You can try, but it'll be a bit of a pain due to how many libs I depend on.
https://github.com/marcmerlin/IoTuz
run examples/fulldemo in it
this includes the 2 files from your code (before your patch) but I did try with your patch and it crashed just the same.
https://github.com/marcmerlin/IoTuz/blob/master/IoTuz.h#L4 define will disable your code and revert to adafruit (to test the code without yours)
https://github.com/marcmerlin/IoTuz/blob/master/examples/fulldemo/fulldemo.ino#L274
remove the ifdef there, since it stopped the crash with one version of esp-idf (but doesn't work on the latest git version), still remove it.
Setup also has a while 1 to remove if the accelerator is not detected.
There may be other things, ping me if you get stuck

MartyMacGyver, 2017-03-23:
If I'm reading the code correctly, `IoTuz::begin()` inits the WS2812 driver, then inits I2C with `accel.begin()`, and that's where it crashes. But I also see that `IoTuz::begin()` calls `Wire.begin()` and then the ADXL345 driver does it again, which seems to be triggering the crash. However, `Wire.begin()` *should* be innocuous.

If the code *as it is on your master branch right now* crashes as-is, let me know. Otherwise, mirror your changes to a branch or attach a zip here with all of it. At the very least I'll be seeing exactly what you're seeing in terms of the running code, even if I can't actually run it effectively.

The fact that this dies when it does is very perplexing. However, if you look at the Arduino-ESP32 I2C HAL, the i2cSetFrequency() code does contain a block bounded by a mutex lock / unlock, which is the only thing I can imagine would cause this:

`#define I2C_MUTEX_LOCK()    do {} while (xSemaphoreTake(i2c->lock, portMAX_DELAY) != pdPASS)`
`#define I2C_MUTEX_UNLOCK()  xSemaphoreGive(i2c->lock)`

Can you also confirm the exact version of Arduino-ESP32 you're using? Just double-checking since the line numbers in the crash dump aren't making much sense.

Finally, right now, does this always crash in the same place?

marcmerlin, 2017-03-23:
First, thank you for having a look. I know it's a lot of code to go through and not too much fun if you don't have the hardware board to see the results :)
You asked which version of arduino esp32 I have. I had an older one from 2 weeks back and I just synced to the git version from @me-no-dev yesterday. It did move my crash around (as in the totally unrelated line of code I had to comment out to work around the bug, now is required to be there for the bug not to trigger)

Ok, https://github.com/marcmerlin/IoTuz has just been updated to be what's on my laptop right now.
adafruit is the default driver (#define NEOPIXEL in IoTuz.h).
The line numbers in github right now should match the exceptions pasted below:

OMG, just removing this single line of code below and my other code changes caused everything to work again with your old and new drivers. There seems to be a variable or code alignment issue. Adding the unrelated line of code fixed your driver now when last week that unrelated line of code caused it to crash.
So I did this (reverted my dirty patch that allowed your code to work before and now causes it to crash):
--- a/examples/fulldemo/fulldemo.ino
+++ b/examples/fulldemo/fulldemo.ino
@@ -245,7 +245,9 @@ void joystick_draw_relative() {
     tft.fillCircle(int(pixel_x), int(pixel_y), 2, tenbitstocolor(update_cnt % 1024));
     // don't go all the way to the border, or the drawing will wrap to the other side of the screen.
     pixel_x = constrain(pixel_x + move_x, 2, 318);
+#ifdef NEOPIXEL
     pixel_y = constrain(pixel_y + move_y, 2, 238);
+#endif

So ifdef'ing out this line of code in my compilation causes the bug to trigger in your code. The same as before, except reversed.
With that line of code removed (NEOPIXEL not defined so that your driver runs instead of adafruit's), here are 3 crashes with my Wire.begin commented out and your old vs new driver:
Note that moving I2C code around (ilke wire.begin, or initializing I2C in another order) causes another bit of I2C to crash. No matter what I do, the first I2C or Wire call I make, crashes.

old driver + double Wire.begin:
Guru Meditation Error of type LoadProhibited occurred on core  1. Exception was unhandled.
Register dump:
PC      : 0x400d889e  PS      : 0x00060230  A0      : 0x800d4f3d  A1      : 0x3ffc7a50  
A2      : 0x3ffc11d0  A3      : 0x00000190  A4      : 0x00000064  A5      : 0x000000c8  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x800d8893  A9      : 0xffffc000  
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x3ffc4134  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0xff000000  SAR     : 0x00000001  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

Backtrace: 0x400d889e:0x3ffc7a50 0x400d4f3d:0x3ffc7a70 0x400d6334:0x3ffc7a90 0x400d4827:0x3ffc7ab0 0x400d1e77:0x3ffc7b00 0x400e5b0a:0x3ffc7b20

0x400d889e: i2cSetFrequency at /home/merlin/Arduino/hardware/espressif/esp32/cores/esp32/esp32-hal-i2c.c line 379
0x400d889e: i2cSetFrequency at /home/merlin/Arduino/hardware/espressif/esp32/cores/esp32/esp32-hal-i2c.c line 379
0x400d4f3d: TwoWire::begin(int, int, unsigned int) at /home/merlin/Arduino/hardware/espressif/esp32/libraries/Wire/src/Wire.cpp line 136
0x400d6334: Adafruit_ADXL345_Unified::begin() at /home/merlin/Arduino/libraries/Adafruit_ADXL345/Adafruit_ADXL345_U.cpp line 291
0x400d4827: IoTuz::begin() at /home/merlin/Arduino/libraries/IoTuz/IoTuz.cpp line 350
0x400d1e77: setup() at /home/merlin/prg/arduino/ESP32/IoTuz/examples/fulldemo/fulldemo.ino line 443
0x400e5b0a: loopTask(void*) at /home/merlin/Arduino/hardware/espressif/esp32/cores/esp32/main.cpp line 17 (discriminator 1)

old driver + comment out first Wire.begin:
Guru Meditation Error of type StoreProhibited occurred on core  1. Exception was unhandled.
Register dump:
PC      : 0x400d89e8  PS      : 0x00060c30  A0      : 0x800d4f44  A1      : 0x3ffc7a40  
A2      : 0x3ffc11d0  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x00000000  
A6      : 0x3ffc11d0  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x3ffc11d0  
A10     : 0x00000001  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0xff000000  SAR     : 0x00000016  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000004  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

Backtrace: 0x400d89e8:0x3ffc7a40 0x400d4f44:0x3ffc7a70 0x400d6324:0x3ffc7a90 0x400d4817:0x3ffc7ab0 0x400d1e77:0x3ffc7b00 0x400e5afa:0x3ffc7b20
0x400d89e8: i2cInit at /home/merlin/Arduino/hardware/espressif/esp32/cores/esp32/esp32-hal-i2c.c line 413
0x400d89e8: i2cInit at /home/merlin/Arduino/hardware/espressif/esp32/cores/esp32/esp32-hal-i2c.c line 413
0x400d4f44: TwoWire::begin(int, int, unsigned int) at /home/merlin/Arduino/hardware/espressif/esp32/libraries/Wire/src/Wire.cpp line 136
0x400d6324: Adafruit_ADXL345_Unified::begin() at /home/merlin/Arduino/libraries/Adafruit_ADXL345/Adafruit_ADXL345_U.cpp line 291
0x400d4817: IoTuz::begin() at /home/merlin/Arduino/libraries/IoTuz/IoTuz.cpp line 350
0x400d1e77: setup() at /home/merlin/prg/arduino/ESP32/IoTuz/examples/fulldemo/fulldemo.ino line 443
0x400e5afa: loopTask(void*) at /home/merlin/Arduino/hardware/espressif/esp32/cores/esp32/main.cpp line 17 (discriminator 1)

new driver + comment out first Wire.begin:
Guru Meditation Error of type StoreProhibited occurred on core  1. Exception was unhandled.
Register dump:
[... cut ...]
0x400d4f40: TwoWire::begin(int, int, unsigned int) at /home/merlin/Arduino/hardware/espressif/esp32/libraries/Wire/src/Wire.cpp line 136
0x400d6320: Adafruit_ADXL345_Unified::begin() at /home/merlin/Arduino/libraries/Adafruit_ADXL345/Adafruit_ADXL345_U.cpp line 291
0x400d4817: IoTuz::begin() at /home/merlin/Arduino/libraries/IoTuz/IoTuz.cpp line 350
0x400d1e77: setup() at /home/merlin/prg/arduino/ESP32/IoTuz/examples/fulldemo/fulldemo.ino line 443
0x400e5af6: loopTask(void*) at /home/merlin/Arduino/hardware/espressif/esp32/cores/esp32/main.cpp line 17 (discriminator 1)

MartyMacGyver, 2017-03-23:
Thank you - I'm examining your code now.

I've cleared out all my local libraries to make this easier, and have the latest Arduino 1.8.2 installed along with the latest (as of yesterday) Arduino-ESP32. I re-ran the "get" step to ensure the compiler set was current.

Since I'm using Windows 7 (without symlink handling enabled) I copied the relevant Handheld files over their symlinks.

I cloned the latest versions of the following libraries (using the default branch for each):
- https://github.com/adafruit/Adafruit-GFX-Library
- https://github.com/adafruit/Adafruit_NeoPixel
- https://github.com/adafruit/Adafruit_Sensor
- https://github.com/adafruit/Adafruit_ADXL345
- https://github.com/PaulStoffregen/XPT2046_Touchscreen
- https://github.com/marcmerlin/Arduino-IRremote
- https://github.com/marcmerlin/Adafruit_ILI9341
- https://github.com/marcmerlin/Adafruit_BME280_Library
- https://github.com/marcmerlin/IoTuz
  -  (Which contains https://github.com/MartyMacGyver/ESP32-Digital-RGB-LED-Drivers already)

I commented out the NEOPIXEL define in ioTuz.h

I was able to build and upload successfully after all that. However, since there is no ADXL345 present I faked it by testing for `if (deviceid == 0xE5)` in `Adafruit_ADXL345_U.cpp` (to avoid removing code or moving lines around).

It goes on to the BME280 check which fails normally. Since this is beyond the points of failure above I'm not going to work around that. Whatever is going on, it's early.

Since you're specifying pins in your code, consider using all of them them in the tft initialization:

IoTuz.h:
	#define TFT_DC 4
	#define TFT_CS 19
	#define TFT_RST 32
	#define SPI_MISO 12
	#define SPI_MOSI 13
	#define SPI_CLK 14

IoTuz.cpp:
	Adafruit_ILI9341 tft = Adafruit_ILI9341(TFT_CS, TFT_DC, SPI_MOSI, SPI_CLK, TFT_RST, SPI_MISO);

I notice you do this in graphicstest.ino (which would be superfluous with this change) but not in anything else.

For this: `SPI.begin(SPI_CLK, SPI_MISO, SPI_MOSI);`, technically you're just using the HSPI pins, so `SPI.begin(-1, -1, 1, -1);` ought to work too. That said, verbosity is better when pin numbers are already being abstracted.

But interestingly, if you aren't using the HSPI pins this call ends up breaking SPI entirely. I tried this with the ESP-WROVER-KIT (which appears to use the VSPI pins) and nada - til I commented that line out. If you delve into `esp32-hal-spi.c` it's not that special yet it has a big effect.

However... the display is being initialized with a bare `begin()` call. If you look at `Adafruit_ILI9341.h:131` this expands to `begin(uint32_t freq = 0, SPIClass &spi=SPI);` specifically for the ESP32. This appears to be passing in a new instance of SPI.

Now let's look at `esp32-hal-spi.c` again. Around line 402 we see a semaphore being created and assigned to `spi->lock`. I looked at this a few times in passing but it's protected by a NULL check so it won't keep creating new semaphores every time it's called... unless it's a new class instance, in which case you get a new semaphore.

Maybe that new semaphore doesn't matter, but this new SPI instance appears to be problematic. At the very least, having two calls to SPI.begin() should work OK, but it seems they do not, at least if you're not using the HSPI pins for either.

Like the Wire.begin() thing I brought up earlier, I wonder how calling SPI.begin() more than once might be having side-effects. The fact that I can't get my TFT to work if that first SPI.begin() call is present points to something - I'm just not sure what that something is yet.

Something is obviously going wrong, and I want my code to work right if got a bug that's causing this. But the way this seems to be moving around still makes me wonder if the problem isn't somewhere else in all this. Maybe try initializing the LEDs after the I2C devices, just to see where it crashes? This would be so much easier if I could replicate this.

**tl;dr - I can't reproduce the actual problem you're reporting, and would be grateful for a greatly simplified example that breaks the board when using my library and which hopefully would have the same effect on my hardware. Multiple instantiations of Wire and SPI may be part of the problem, and race conditions and/or memory corruption have not been ruled out (involving WIre, SPI, or my library.**

marcmerlin, 2017-03-24:
TL;DR: I removed the double Wire.begin call, I removed the double SPI.begin() call and I moved your
LED init after the I2C init, and in all cases I still got the crash :(
More details below:

First, thank you for your patience with trying all this and good job getting all
the relevant pieces, especially the ILI lib from my github (I need to update the
readme to point to that one if adafruit isn't goint to merge my latest change).

As for your request "please get this crashing in a smaller example", yes, I really tried. I took your demo
code and added all my fulldemo.ino code to it block by block until all of it was there, and it didn't crash.
Then I realized that even my own code doesn't crash anymore unless I ifdef out that magic line:
#ifdef NEOPIXEL
    pixel_y = constrain(pixel_y + move_y, 2, 238);
#endif
that was the line which I had to remove for your lib to work, and not I have to not remove it for the lib to work.
Needless to say that this is the most WTF bug I've ever seen so far :-/

your other comments:
- https://github.com/marcmerlin/Adafruit_BME280_Library
No idea why I have this fork, it's unmodified anyway so no biggie, I'll delete it

- "Since you're specifying pins in your code, consider using all of them them in the tft initialization"
tried, but no change crash-wise.
However, this causes my screen to sometimes init white and not display anything and I don't now why, so
I went back from
Adafruit_ILI9341 tft = Adafruit_ILI9341(TFT_CS, TFT_DC, SPI_MOSI, SPI_CLK, TFT_RST, SPI_MISO);
to
Adafruit_ILI9341 tft = Adafruit_ILI9341(TFT_CS, TFT_DC, TFT_RST);

"For this: SPI.begin(SPI_CLK, SPI_MISO, SPI_MOSI);, technically you're just using the HSPI pins,"
I didn't even know that (I didn't do the board or pin mapping). I didn't know there were HSPI pins by default. I thought the assignment was purely virtual and it didn't matter which pins you take. But as you said, it shouldn't make a difference.

About 2 SPI.Begin() calls, you are correct, although I've been told the first one wins, and adafruit in their other libs also recommend to do your own SPI.begin on ESP8266/ESP32 before calling their lib, so I hope it's ok?
If I remove my SPI.begin, the TFT library still works but the touchscreen SPI library stops working. But eh, at least the screen turns on. Then, if I
do this:
 //   pixel_y = constrain(pixel_y + move_y, 2, 238);
then the code crashes as usual
0x400d853b: i2cResetFiFo at /home/merlin/Arduino/hardware/espressif/esp32/cores/esp32/esp32-hal-i2c.c line 379
0x400d853b: i2cResetFiFo at /home/merlin/Arduino/hardware/espressif/esp32/cores/esp32/esp32-hal-i2c.c line 379
0x400d85d3: i2cWrite at /home/merlin/Arduino/hardware/espressif/esp32/cores/esp32/esp32-hal-i2c.c line 379
0x400d5094: TwoWire::endTransmission(unsigned char) at /home/merlin/Arduino/hardware/espressif/esp32/libraries/Wire/src/Wire.cpp line 136
0x400d50e0: TwoWire::endTransmission() at /home/merlin/Arduino/hardware/espressif/esp32/libraries/Wire/src/Wire.cpp line 136
0x400d46c1: IoTuz::pcf8574_write_(unsigned char) at /home/merlin/Arduino/libraries/IoTuz/IoTuz.cpp line 110

Initializing the LEDs after I2C: I did try that earlier. If I do that, the first call to Wire crashes, and I call Wire often to interrogate my I/O expander and read its pins via I2C.

IoTuz Setup done
Selection Box Size: 64 x 48
Turn backlight on
Guru Meditation Error of type LoadProhibited occurred on core  1. Exception was unhandled.
0x400d854b: i2cResetFiFo at /home/merlin/Arduino/hardware/espressif/esp32/cores/esp32/esp32-hal-i2c.c line 379
0x400d854b: i2cResetFiFo at /home/merlin/Arduino/hardware/espressif/esp32/cores/esp32/esp32-hal-i2c.c line 379
0x400d85e3: i2cWrite at /home/merlin/Arduino/hardware/espressif/esp32/cores/esp32/esp32-hal-i2c.c line 379
0x400d50a4: TwoWire::endTransmission(unsigned char) at /home/merlin/Arduino/hardware/espressif/esp32/libraries/Wire/src/Wire.cpp line 136
0x400d50f0: TwoWire::endTransmission() at /home/merlin/Arduino/hardware/espressif/esp32/libraries/Wire/src/Wire.cpp line 136
0x400d46c1: IoTuz::pcf8574_write_(unsigned char) at /home/merlin/Arduino/libraries/IoTuz/IoTuz.cpp line 110
0x400d46df: IoTuz::i2cexp_clear_bits(unsigned char) at /home/merlin/Arduino/libraries/IoTuz/IoTuz.cpp line 121
0x400d470d: IoTuz::screen_bl(bool) at /home/merlin/Arduino/libraries/IoTuz/IoTuz.cpp line 181 (discriminator 1)
0x400d1e6e: setup() at /home/merlin/prg/arduino/ESP32/IoTuz/examples/fulldemo/fulldemo.ino line 445
0x400e5c06: loopTask(void*) at /home/merlin/Arduino/hardware/espressif/esp32/cores/esp32/main.cpp line 17 (discriminator 1)

Ok, if I comment out this call:
    //iotuz.screen_bl(true);
then it dies on the first call to the IO expander:
Logo displayed, waiting for timeout or click
Guru Meditation Error of type LoadProhibited occurred on core  1. Exception was unhandled.

CPU halted.
0x400d8543: i2cResetFiFo at /home/merlin/Arduino/hardware/espressif/esp32/cores/esp32/esp32-hal-i2c.c line 379
0x400d8543: i2cResetFiFo at /home/merlin/Arduino/hardware/espressif/esp32/cores/esp32/esp32-hal-i2c.c line 379
0x400d8782: i2cRead at /home/merlin/Arduino/hardware/espressif/esp32/cores/esp32/esp32-hal-i2c.c line 379
0x400d5062: TwoWire::requestFrom(unsigned char, unsigned int, bool) at /home/merlin/Arduino/hardware/espressif/esp32/libraries/Wire/src/Wire.cpp line 136
0x400d50d4: TwoWire::requestFrom(int, int) at /home/merlin/Arduino/hardware/espressif/esp32/libraries/Wire/src/Wire.cpp line 136
0x400d4a8e: IoTuz::i2cexp_read() at /home/merlin/Arduino/libraries/IoTuz/IoTuz.cpp line 135
0x400d4ab0: IoTuz::read_encoder_button() at /home/merlin/Arduino/libraries/IoTuz/IoTuz.cpp line 164
0x400d1ebf: setup() at /home/merlin/prg/arduino/ESP32/IoTuz/examples/fulldemo/fulldemo.ino line 445
0x400e5bfe: loopTask(void*) at /home/merlin/Arduino/hardware/espressif/esp32/cores/esp32/main.cpp line 17 (discriminator 1)

marcmerlin, 2017-03-24:
So, I commented out all my I2C code to see how long things would run, and then got:
0x400d842f: i2cResetFiFo at /home/merlin/Arduino/hardware/espressif/esp32/cores/esp32/esp32-hal-i2c.c line 379
0x400d584c: Adafruit_ILI9341::endWrite() at /home/merlin/Arduino/libraries/Adafruit_ILI9341/Adafruit_ILI9341.cpp line 498
0x400d842f: i2cResetFiFo at /home/merlin/Arduino/hardware/espressif/esp32/cores/esp32/esp32-hal-i2c.c line 379
0x400d84c7: i2cWrite at /home/merlin/Arduino/hardware/espressif/esp32/cores/esp32/esp32-hal-i2c.c line 379
0x400d4fe0: TwoWire::endTransmission(unsigned char) at /home/merlin/Arduino/hardware/espressif/esp32/libraries/Wire/src/Wire.cpp line 136
0x400d502c: TwoWire::endTransmission() at /home/merlin/Arduino/hardware/espressif/esp32/libraries/Wire/src/Wire.cpp line 136
0x400d69dc: Adafruit_BME280::read24(unsigned char) at /home/merlin/Arduino/libraries/Adafruit_BME280_Library/Adafruit_BME280.cpp line 273
0x400d6a80: Adafruit_BME280::readTemperature() at /home/merlin/Arduino/libraries/Adafruit_BME280_Library/Adafruit_BME280.cpp line 375
0x400d14df: HumiTemp_Handler() at /home/merlin/prg/arduino/ESP32/IoTuz/examples/fulldemo/fulldemo.ino line 445

Ok, I removed that too, and then no crash anymore.
So basically your code runs fine, even when I'm in the configuration where it crashes _as long as I don't make any i2c call at all_.
But then again by moving code around, like you already know now, I can get I2C + your code to work forever (seemingly) too. 
Sadly, this is also why it's so hard for me to give you a short piece of code that crashes.

On the plus side, I just confirmed that if I'm in a crashing configuration, not doing any I2C whatsoever after initializing your code, makes crashes go away and I can change LED colors forever without issues. 
Hope this helps just a tiny bit.

MartyMacGyver, 2017-03-24:
Really wish I could see this crash. I have a suspicion that this is related to HSPI if only because it seems to be a part of he problem.... what I basically need is a crashing configuration, and it's easier if I have just a display, an I2C device and a few LEDs.

Since your baro is crashing it now I can probably get away with testing against my MS5637 + another 9341 LCD I happen to have + some 2812s.

Can you branch and upload your stripped down but still crashing code? Pull out as much else as you can, as well as any unused includes. Since I'm flying blind this will reduce iteration time. It needn't be fancy - start the LCD, init the LEDs and then try to read from the I2C baro. If it needs more than that to crash, let me know. If you can get it to crash without the LCD code, let me know - that would simplify this greatly.

For my part I will work to mirror your pin-level config using my Core Board v2 and the parts I mentioned. We see already that the pins you use can silently influence the choices made in the hal for SPI - perhaps this is central to the problem for I2C as well. I really want to get this resolved and being able to replicate it will help greatly.

MartyMacGyver, 2017-03-24:
Note: I don't expect short code, but shorter. Eliminate anything after the crash and anything else not necessary to make it crash. Right now you can iterate on that far more effectively because you see the crash... And hopefully a reduced footprint will make it more.likely to crash for me too.

I appreciate your help in this - whatever the problem turns out to be, it will be a useful problem to resolve.

marcmerlin, 2017-03-24:
Yes, I understand what you mean, let me work on this after dinner, and thanks to you for your time on this too (and yes, hopefully it will be something interesting :) )

MartyMacGyver, 2017-03-24:
I'll be eager to see it when I get back (Meetup + networking for the next couple of hours) or later tonight.

marcmerlin, 2017-03-24:
Ok, that did take 2 hours of work, but here is a much smaller version.
It crashes like so:
IoTuz Setup done
Selection Box Size: 64 x 48
Turn backlight on
Guru Meditation Error of type LoadProhibited occurred on core  1. Exception was unhandled.
Register dump:
0x400d3333: i2cResetFiFo at /home/merlin/Arduino/hardware/espressif/esp32/cores/esp32/esp32-hal-i2c.c line 379
0x400d3333: i2cResetFiFo at /home/merlin/Arduino/hardware/espressif/esp32/cores/esp32/esp32-hal-i2c.c line 379
0x400d33cb: i2cWrite at /home/merlin/Arduino/hardware/espressif/esp32/cores/esp32/esp32-hal-i2c.c line 379
0x400d1b6c: TwoWire::endTransmission(unsigned char) at /home/merlin/Arduino/hardware/espressif/esp32/libraries/Wire/src/Wire.cpp line 143
0x400d1ba0: TwoWire::endTransmission() at /home/merlin/Arduino/hardware/espressif/esp32/libraries/Wire/src/Wire.cpp line 143
0x400d1385: IoTuz::pcf8574_write_(unsigned char) at /home/merlin/Arduino/libraries/IoTuz/IoTuz.cpp line 59
0x400d13a3: IoTuz::i2cexp_clear_bits(unsigned char) at /home/merlin/Arduino/libraries/IoTuz/IoTuz.cpp line 70
0x400d13d1: IoTuz::screen_bl(bool) at /home/merlin/Arduino/libraries/IoTuz/IoTuz.cpp line 94 (discriminator 1)
0x400d0c9e: setup() at /home/merlin/arduino/libraries/IoTuz/examples/fulldemo/fulldemo.ino line 303
0x400df5ba: loopTask(void*) at /home/merlin/Arduino/hardware/espressif/esp32/cores/esp32/main.cpp line 17 (discriminator 1)

However if I take away the accelerator code, then it doesn't crash anymore.
Hopefully this is helpful, it took a lot of effort to gut my code like this ;)
But honestly, I'm not hopeful it will work/crash for you. If I remove the rgbLedFade function (function that never get called), the code doesn't crash anymore :(
[iotuz1.tar.gz](https://github.com/MartyMacGyver/ESP32-Digital-RGB-LED-Drivers/files/866924/iotuz1.tar.gz)

hope this helps, somehow. If it works for you, it's a smaller codebase to start hacking/verifying from

MartyMacGyver, 2017-03-24:
It doesn't crash for me... yet. I see the LCD code is still in there - is this or any of the SPI stuff necessary to lead to a crash?

MartyMacGyver, 2017-03-24:
Also, given the exact code you just sent me, I want you to do the following: build the code using Arduino as you normally do, and attach a tarball with the build output (include a copy of the build log too). That build output folder will include files like `arduino.ar` and `fulldemo.ino.bin` and lots of other stuff. I want to see how my build artifacts compare, as well as to see what flashing the exact same binary outputs to my board accomplishes. If it doesn't crash (or if the artifacts match), that points me in a certain direction.

By the way, have you cleared your nvram on the board? May not have any effect on the problem but it might be good to do that, then re-flash the software to verify it's still acting up.

`esptool.py --port <serial_port> erase_flash`

marcmerlin, 2017-03-24:
[build2.tar.gz](https://github.com/MartyMacGyver/ESP32-Digital-RGB-LED-Drivers/files/868426/build2.tar.gz)

[iotuz2.tar.gz](https://github.com/MartyMacGyver/ESP32-Digital-RGB-LED-Drivers/files/868424/iotuz2.tar.gz)

1) did ./esptool --port /dev/ttyUSB0 erase_flash
didn't help.
2) removed SPI, still crashes. Here's the smaller source and arduino build bin

Crash:
Turn backlight on
Guru Meditation Error of type LoadProhibited occurred on core  1. Exception was unhandled.
0x40080000: _WindowOverflow4 at ?? line ?
0x40080034: _WindowOverflow4 at ?? line ?
0x400d193f: i2cSetCmd at /home/merlin/Arduino/hardware/espressif/esp32/cores/esp32/esp32-hal-i2c.c line 379
0x400d193f: i2cSetCmd at /home/merlin/Arduino/hardware/espressif/esp32/cores/esp32/esp32-hal-i2c.c line 379
0x400d19d7: i2cResetFiFo at /home/merlin/Arduino/hardware/espressif/esp32/cores/esp32/esp32-hal-i2c.c line 379
0x400d11e4: TwoWire::endTransmission(unsigned char) at /home/merlin/Arduino/hardware/espressif/esp32/libraries/Wire/src/Wire.cpp line 143
0x400d1218: TwoWire::requestFrom(int, int) at /home/merlin/Arduino/hardware/espressif/esp32/libraries/Wire/src/Wire.cpp line 143
0x400d0b41: IoTuz::pcf8574_write_(unsigned char) at /home/merlin/Arduino/libraries/IoTuz/IoTuz.cpp line 89
0x400d0b5f: IoTuz::i2cexp_clear_bits(unsigned char) at /home/merlin/Arduino/libraries/IoTuz/IoTuz.cpp line 89
0x400d0b8d: IoTuz::screen_bl(bool) at /home/merlin/Arduino/libraries/IoTuz/IoTuz.cpp line 89
0x400d0976: setup() at /home/merlin/arduino/libraries/IoTuz/examples/fulldemo/fulldemo.ino line 440
0x400d99ce: esp_vfs_rename at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/vfs/./vfs.c line 331

build log:
Detecting libraries used...
"/home/merlin/Arduino/hardware/espressif/esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM -DMBEDTLS_CONFIG_FILE="mbedtls/esp_config.h" -DHAVE_CONFIG_H "-I/home/merlin/Arduino/hardware/espressif/esp32/tools/sdk/include/config" "-I/home/merlin/Arduino/hardware/espressif/esp32/tools/sdk/include/bluedroid" "-I/home/merlin/Arduino/hardware/espressif/esp32/tools/sdk/include/bootloader_support" "-I/home/merlin/Arduino/hardware/espressif/esp32/tools/sdk/include/bt" "-I/home/merlin/Arduino/hardware/espressif/esp32/tools/sdk/include/driver" "-I/home/merlin/Arduino/hardware/espressif/esp32/tools/sdk/include/esp32" "-I/home/merlin/Arduino/hardware/espressif/esp32/tools/sdk/include/ethernet" "-I/home/merlin/Arduino/hardware/espressif/esp32/tools/sdk/include/fatfs" "-I/home/merlin/Arduino/hardware/espressif/esp32/tools/sdk/include/freertos" "-I/home/merlin/Arduino/hardware/espressif/esp32/tools/sdk/include/log" "-I/home/merlin/Arduino/hardware/espressif/esp32/tools/sdk/include/mdns" "-I/home/merlin/Arduino/hardware/espressif/esp32/tools/sdk/include/mbedtls" "-I/home/merlin/Arduino/hardware/espressif/esp32/tools/sdk/include/mbedtls_port" "-I/home/merlin/Arduino/hardware/espressif/esp32/tools/sdk/include/vfs" "-I/home/merlin/Arduino/hardware/espressif/esp32/tools/sdk/include/ulp" "-I/home/merlin/Arduino/hardware/espressif/esp32/tools/sdk/include/newlib" "-I/home/merlin/Arduino/hardware/espressif/esp32/tools/sdk/include/nvs_flash" "-I/home/merlin/Arduino/hardware/espressif/esp32/tools/sdk/include/spi_flash" "-I/home/merlin/Arduino/hardware/espressif/esp32/tools/sdk/include/sdmmc" "-I/home/merlin/Arduino/hardware/espressif/esp32/tools/sdk/include/openssl" "-I/home/merlin/Arduino/hardware/espressif/esp32/tools/sdk/include/app_update" "-I/home/merlin/Arduino/hardware/espressif/esp32/tools/sdk/include/tcpip_adapter" "-I/home/merlin/Arduino/hardware/espressif/esp32/tools/sdk/include/xtensa-debug-module" "-I/home/merlin/Arduino/hardware/espressif/esp32/tools/sdk/include/newlib" "-I/home/merlin/Arduino/hardware/espressif/esp32/tools/sdk/include/coap" "-I/home/merlin/Arduino/hardware/espressif/esp32/tools/sdk/include/wpa_supplicant" "-I/home/merlin/Arduino/hardware/espressif/esp32/tools/sdk/include/expat" "-I/home/merlin/Arduino/hardware/espressif/esp32/tools/sdk/include/json" "-I/home/merlin/Arduino/hardware/espressif/esp32/tools/sdk/include/nghttp" "-I/home/merlin/Arduino/hardware/espressif/esp32/tools/sdk/include/lwip" -std=gnu++11 -fno-exceptions -fno-rtti -Os -g3 -Wpointer-arith -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -mlongcalls -nostdlib -w -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wno-unused-parameter -Wno-sign-compare -fno-rtti  -c -w -x c++ -E -CC -DF_CPU=240000000L -DARDUINO=10801 -DARDUINO_ESP32_DEV -DARDUINO_ARCH_ESP32  -DESP32 -DCORE_DEBUG_LEVEL=0 "-I/home/merlin/Arduino/hardware/espressif/esp32/cores/esp32" "-I/home/merlin/Arduino/hardware/espressif/esp32/variants/esp32" "/tmp/arduino_build_812614/sketch/fulldemo.ino.cpp" -o "/dev/null"
"/home/merlin/Arduino/hardware/espressif/esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM -DMBEDTLS_CONFIG_FILE="mbedtls/esp_config.h" -DHAVE_CONFIG_H "-I/home/merlin/Arduino/hardware/espressif/esp32/tools/sdk/include/config" "-I/home/merlin/Arduino/hardware/espressif/esp32/tools/sdk/include/bluedroid" "-I/home/merlin/Arduino/hardware/espressif/esp32/tools/sdk/include/bootloader_support" "-I/home/merlin/Arduino/hardware/espressif/esp32/tools/sdk/include/bt" "-I/home/merlin/Arduino/hardware/espressif/esp32/tools/sdk/include/driver" "-I/home/merlin/Arduino/hardware/espressif/esp32/tools/sdk/include/esp32" "-I/home/merlin/Arduino/hardware/espressif/esp [... cut ...]
Hash of data verified.
Compressed 8192 bytes to 47...

Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.0 seconds (effective 35313.2 kbit/s)...
Hash of data verified.
Compressed 198208 bytes to 69742...

Writing at 0x00010000... (20 %)
Writing at 0x00014000... (40 %)
Writing at 0x00018000... (60 %)
Writing at 0x0001c000... (80 %)
Writing at 0x00020000... (100 %)
Wrote 198208 bytes (69742 compressed) at 0x00010000 in 2.0 seconds (effective 792.8 kbit/s)...
Hash of data verified.

MartyMacGyver, 2017-03-24:
Thank you!

I have a question - how are you getting such verbose stack traces? Mine never show the code like yours do...

marcmerlin, 2017-03-24:
You mean this?
https://github.com/marcmerlin/IoTuz/blob/master/README.md
"Please make sure you setup the exception decoder, it will make your life much easier when you get a crash dump on your serial port:

https://github.com/espressif/arduino-esp32#installation-instructions
https://github.com/me-no-dev/EspExceptionDecoder <<< this"

marcmerlin, 2017-03-24:
note that every time I get a numerical stack strace, I paste it in the decoder and give you the decoded version instead

MartyMacGyver, 2017-03-24:
Yeah, that... (I didn't know you had it in your readme). Handy tool - added to my notes and installation.

In related news... your binary crashes just fine - and now, so does mine.

Flash process used:

	esptool.py \
	  --chip esp32 \
	  --port COM17 \
	  --baud 921600 \
	  --before default_reset \
	  --after hard_reset write_flash -z \
	  --flash_mode dio \
	  --flash_freq 80m \
	  --flash_size detect 0x1000 \
	  /T/Projects/Arduino/_workspace/hardware/espressif/esp32/tools/sdk/bin/bootloader.bin 0x8000 \
	  fulldemo.ino.partitions.bin 0xe000 \
	  /T/Projects/Arduino/_workspace/hardware/espressif/esp32/tools/partitions/boot_app0.bin 0x10000 \
	  fulldemo.ino.bin 

My crash:

	ets Jun  8 2016 00:22:57

	rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
	configsip: 0, SPIWP:0x00
	clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
	mode:DIO, clock div:1
	load:0x3fff0008,len:8
	load:0x3fff0010,len:2016
	load:0x40078000,len:7780
	ho 0 tail 12 room 4
	load:0x40080000,len:252
	entry 0x40080034
	Serial Begin
	FF
	Ooops, no ADXL345 detected ... Check your wiring!
	Turning on LEDs
	Before colors
	After colors
	IoTuz Setup done
	Turn backlight on
	Guru Meditation Error of type LoadProhibited occurred on core  1. Exception was unhandled.
	Register dump:
	PC      : 0x400d19b7  PS      : 0x00060630  A0      : 0x800d1a4f  A1      : 0x3ffc73a0  
	A2      : 0x3ffc11d0  A3      : 0x00000000  A4      : 0x3ffc78bc  A5      : 0x00000000  
	A6      : 0x00000000  A7      : 0x00060023  A8      : 0x00002000  A9      : 0x00000000  
	A10     : 0x3ffc78bc  A11     : 0x00000000  A12     : 0x00060023  A13     : 0x3ffc7564  
	A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x0000001f  EXCCAUSE: 0x0000001c  
	EXCVADDR: 0x00000018  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

	Backtrace: 0x400d19b7:0x3ffc73a0 0x400d1a4f:0x3ffc73c0 0x400d11f4:0x3ffc7410 0x400d1228:0x3ffc7430 0x400d0b51:0x3ffc7450 0x400d0b6f:0x3ffc7470 0x400d0b9d:0x3ffc7490 0x400d0986:0x3ffc74b0 0x400d9a46:0x3ffc74d0

	CPU halted.

	Decoding 15 results
	0x40080000: _WindowOverflow4 at ?? line ?
	0x40080034: _WindowOverflow4 at ?? line ?
	0x400d19b7: i2cResetFiFo at T:\Projects\Arduino\_workspace\hardware\espressif\esp32\cores\esp32/esp32-hal-i2c.c line 379
	0x400d19b7: i2cResetFiFo at T:\Projects\Arduino\_workspace\hardware\espressif\esp32\cores\esp32/esp32-hal-i2c.c line 379
	0x400d1a4f: i2cWrite at T:\Projects\Arduino\_workspace\hardware\espressif\esp32\cores\esp32/esp32-hal-i2c.c line 379
	0x400d11f4: TwoWire::endTransmission(unsigned char) at T:\Projects\Arduino\_workspace\hardware\espressif\esp32\libraries\Wire\src/Wire.cpp line 143
	0x400d1228: TwoWire::endTransmission() at T:\Projects\Arduino\_workspace\hardware\espressif\esp32\libraries\Wire\src/Wire.cpp line 143
	0x400d0b51: IoTuz::pcf8574_write_(unsigned char) at T:\Projects\Arduino\_workspace\libraries\IoTuz/IoTuz.cpp line 89
	0x400d0b6f: IoTuz::i2cexp_clear_bits(unsigned char) at T:\Projects\Arduino\_workspace\libraries\IoTuz/IoTuz.cpp line 89
	0x400d0b9d: IoTuz::screen_bl(bool) at T:\Projects\Arduino\_workspace\libraries\IoTuz/IoTuz.cpp line 89
	0x400d0986: setup() at T:\Projects\Arduino\_workspace\libraries\IoTuz\examples\fulldemo/fulldemo.ino line 440
	0x400d9a46: loopTask(void*) at T:\Projects\Arduino\_workspace\hardware\espressif\esp32\cores\esp32/main.cpp line 17 (discriminator 1)

I made a small mistake last night with `iotuz1.tar.gz` - I used your pre-strip library files with the stripped code and as we saw it refused to crash (as it refused to crash with your original code). Using `iotuz2.tar.gz` as both library and example is crashing from a clean build here, and in the same way, so at least we can finally see the same problem and more importantly, I can replicate it without extra hardware.

Now I can delve into this and narrow down what's going wrong.

marcmerlin, 2017-03-24:
Ah, this is awesome news. I was worried that after all those hours I spent stripping the code (>4) it would still work for you :)
Thanks again for having a look, and good luck finding the issue.

and yes, the exception decoder is awesome, I learned about it from @projectgus at linux.conf.au hardware conf where we built the IoTuz board I wrote this code for. Those decoded tracebacks are a huge plus to figuring out WTH went wrong and where :)

marcmerlin, 2017-03-24:
Eh, a dumb question (maybe?).  Is there just one RMT in the chip?
If I use your code for LEDs, does it stop me from also using RMT for IR remote control as per this lib?
https://github.com/ExploreEmbedded/ESP32_RMT
For now, I just made the interrupt driven sparkfun lib work on ESP32 instead: https://github.com/z3t0/Arduino-IRremote/pull/425

MartyMacGyver, 2017-03-24:
You mentioned `constrain()` early on, and it's coming back around to that. For reference, here's the macro definition in `cores/esp32/Arduino.h`:

`#define constrain(amt,low,high) ((amt)<(low)?(low):((amt)>(high)?(high):(amt)))`

Pretty simple really.

I started chopping out unused blocks of code (very little is needed in `loop()` at this point) and removed `rgbLedFade()`... suddenly it all worked again. Odd. I narrowed it down to... the `constrain()` call. I reverted the block, and it crashes. I unwound the `constrain()` call into a full expansion and it still crashes (seems obvious, but at this point I'm checking everything).

The upshot? _That function `rgbLedFade()` isn't even used._

Which is basically how you described this early on, before I tried to replicate all this.

I should be able to put together a MUCH smaller example to test against now (or condense yours). I'm starting to think it might actually be a compiler or alignment bug after all, but it's a little early to say that.

I'm doing all this in a small local git repo so I can track my progress, from the `iotuz2.tar.gz` as basis to whatever I end up with.

marcmerlin, 2017-03-24:
Yeah, rgbLedFade() isn't even used and yet when I removed it, the bug went away.
that's why I'm still wondering if it is a compiler/alignment bug. The fact that removing code that never gets run changes the binary in a way that it works, or not, is just the weirdest thing.
By the way, I don't think it's constrain at all. At some point, I just replaced it y = y +1 and it still crashed, but if I removed the line, the crash went away

MartyMacGyver, 2017-03-24:
My concern is that something is walking over memory somewhere and this is just a symptom.

Could be an optimization bug too... in a previous job I would occasionally run into very weird optimization-related bugs. This Arduino build uses -Os. From the man page:

	-Os
	Optimize for size. -Os enables all -O2 optimizations that do not typically increase code size. It also performs further optimizations designed to reduce code size.

	-Os disables the following optimization flags:

	-falign-functions  -falign-jumps  -falign-loops 
	-falign-labels  -freorder-blocks  -freorder-blocks-algorithm=stc 
	-freorder-blocks-and-partition  -fprefetch-loop-arrays

I'd be curious to see what effect -O0 has... might have to give that a try once I reduce the code set some more.

MartyMacGyver, 2017-03-24:
And the winner is... optimization.

I edited my `hardware/espressif/esp32/platform.txt`and changed the two instances of -Os to -O0. Rebuilt (do a clean build - remove temp files first (Windows does this automagically, but I'm not sure Linux/Mac builds do)) and try it out. If that works, try your full build out.

This isn't really a solution, but it's definitely a direction to go in.

I'm going to look into what version of GCC this toolchain is based on.

marcmerlin, 2017-03-24:
@me-no-dev @projectgus you may be interested in what Marty just found (thanks for that by the way), apparently indeed a compiler bug related to optimizations.

MartyMacGyver, 2017-03-24:
Well, I went back to your original stripped version and it's crashing again so I don't know... but optimization definitely had an impact. I am more suspicious of the compiler now, but I still need to reduce the problem space.

me-no-dev, 2017-03-24:
I'm following your conversation :) I'm sure @projectgus does too.
BTW I will look into the compiler flags ;)

MartyMacGyver, 2017-03-24:
@me-no-dev Don't go too deep into that just yet. I'm stripping down the code now and will hopefully have a smaller dataset to test against tonight.

What's really weird (or telling) is how the problem comes and goes depending on what code you change. I removed an early definition of a multi-dimensional string array (unused) and the crash went away. `constrain()` is a bit complicated but it seems much more like this is an alignment issue or some kind of data overwrite thing. The fact that disabling optimization changes the effect is interesting, but it doesn't eliminate the crash in all cases so it's probably not the root cause (which is something of a relief).

MartyMacGyver, 2017-03-25:
[IoTuz_wip.tar.gz](https://github.com/MartyMacGyver/ESP32-Digital-RGB-LED-Drivers/files/869615/IoTuz_wip.tar.gz)

Attached is a greatly reduced example that demonstrates the problem. If `ire.begin()` happens before `ws2812_setColors()`, a subsequent call to `Wire.endTransmission()` will cause a crash. However, if the little block of `constrain()`-using code in loop() (which comes after all that) is removed... it works fine.

@marcmerlin If you get a moment, build this code as-is and see if it causes a crash for you too. It should. Then remove the constrain() block as noted here and in the code and let me know if it works again.

The crash is slightly different now, though strategic placement of extra superfluous blocks of code can change the characteristics of the error slightly it's always connected to i2cWrite() in the hal.

	ets Jun  8 2016 00:22:57

	rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
	configsip: 0, SPIWP:0x00
	clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
	mode:DIO, clock div:1
	load:0x3fff0008,len:8
	load:0x3fff0010,len:2016
	load:0x40078000,len:7780
	ho 0 tail 12 room 4
	load:0x40080000,len:252
	entry 0x40080034
	Serial Begin
	Talking to I2C device
	  -- Wire.begin()
	Turn on LEDs
	Talking to I2C device (continued)
	  -- Begin transmission
	  -- Write data
	  -- End transmission
	/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./queue.c:1444 (xQueueGenericReceive)- assert failed!
	abort() was called at PC 0x40083759
	Guru Meditation Error: Core  1 panic'ed (abort)

	Backtrace: 0x40008155:0x3ffc72a0 0x40007d16:0x3ffc72c0 0x400d10dc:0x3ffc7300 0x400d0e18:0x3ffc7350 0x400d0e4c:0x3ffc7370 0x400d07be:0x3ffc7390 0x400d8f9a:0x3ffc73b0

	CPU halted.

	Decoding 11 results
	0x40080000: _WindowOverflow4 at ?? line ?
	0x40080034: _WindowOverflow4 at ?? line ?
	0x40083759: xQueueGenericReceive at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./queue.c line 1998
	0x400d10dc: i2cWrite at hardware\espressif\esp32\cores\esp32/esp32-hal-i2c.c line 379
	0x400d0e18: TwoWire::endTransmission(unsigned char) at hardware\espressif\esp32\libraries\Wire\src/Wire.cpp line 143
	0x400d0e4c: TwoWire::endTransmission() at hardware\espressif\esp32\libraries\Wire\src/Wire.cpp line 143
	0x400d07be: setup() at IoTuz_wip\fulldemo/fulldemo.ino line 97
	0x400d8f9a: loopTask(void*) at hardware\espressif\esp32\cores\esp32/main.cpp line 17 (discriminator 1)

At this point it's either a compiler bug, a code bug, or an interaction with system semaphores.

marcmerlin, 2017-03-25:
@MartyMacGyver good job taking the code way down still. Yes, I can confirm it crashes for me still.
Note that as said earlier, you don't need constrain. It crashes with this:
      zzza = zzza+1;
      zzzb = zzzb+1;
Then, you can make the code work by simply commenting out the single line:
      zzzb = zzzb+1;

MartyMacGyver, 2017-03-25:
I kept sprinkling in `constrain()` because it was quick and effective. I kind of doubt it's a problem itself.

I decided to create a crashing and non-crashing build and disassemble each with `xtensa-esp32-elf-objdump.exe -d fulldemo.ino.elf`. The results are attached:

[good_and_bad.tar.gz](https://github.com/MartyMacGyver/ESP32-Digital-RGB-LED-Drivers/files/869986/good_and_bad.tar.gz)

Here's what I changed, along with the relevant dump:

# Version 1 (latest version from earlier upload, crashes)

Relevant code:

	void loop() {
	    Serial.println("Constrain something");

	    // Remove this block though, and it all works again...
	    {
	      static float zzza = 5;
	      static float zzzb = 5;
	      zzza = constrain(zzza, 1, 9);
	      zzzb = constrain(zzzb, 1, 9);
	    }
	    
	    Serial.println("Constraint complete");
	    Serial.println("Halting");
	    while(1) {}
}

Disassembly of `loop()`:

	400d0690 <_Z4loopv>:
	400d0690:	004136        	entry	a1, 32
	400d0693:	fe6421        	l32r	a2, 400d0024 <_stext+0xc>
	400d0696:	fe62b1        	l32r	a11, 400d0020 <_stext+0x8>
	400d0699:	02ad      	mov.n	a10, a2
	400d069b:	0127a5        	call8	400d1914 <_ZN5Print7printlnEPKc>
	400d069e:	fe6281        	l32r	a8, 400d0028 <_stext+0x10>
	400d06a1:	fe5d31        	l32r	a3, 400d0018 <_stext>
	400d06a4:	000813        	lsi	f1, a8, 0
	400d06a7:	fa0350        	wfr	f0, a3
	400d06aa:	4b0100        	olt.s	b0, f1, f0
	400d06ad:	111076        	bt	b0, 400d06c2 <_Z4loopv+0x32>
	400d06b0:	fe5b31        	l32r	a3, 400d001c <_stext+0x4>
	400d06b3:	fa2350        	wfr	f2, a3
	400d06b6:	4b0210        	olt.s	b0, f2, f1
	400d06b9:	0b1076        	bt	b0, 400d06c8 <_Z4loopv+0x38>
	400d06bc:	0002c6        	j	400d06cb <_Z4loopv+0x3b>
	400d06bf:	000000        	ill
	400d06c2:	fa1350        	wfr	f1, a3
	400d06c5:	000086        	j	400d06cb <_Z4loopv+0x3b>
	400d06c8:	fa1350        	wfr	f1, a3
	400d06cb:	004813        	ssi	f1, a8, 0
	400d06ce:	fe5781        	l32r	a8, 400d002c <_stext+0x14>
	400d06d1:	000813        	lsi	f1, a8, 0
	400d06d4:	4b0100        	olt.s	b0, f1, f0
	400d06d7:	0e1076        	bt	b0, 400d06e9 <_Z4loopv+0x59>
	400d06da:	fe5031        	l32r	a3, 400d001c <_stext+0x4>
	400d06dd:	fa0350        	wfr	f0, a3
	400d06e0:	4b0010        	olt.s	b0, f0, f1
	400d06e3:	021076        	bt	b0, 400d06e9 <_Z4loopv+0x59>
	400d06e6:	000086        	j	400d06ec <_Z4loopv+0x5c>
	400d06e9:	fa1000        	mov.s	f1, f0
	400d06ec:	fe51b1        	l32r	a11, 400d0030 <_stext+0x18>
	400d06ef:	20a220        	or	a10, a2, a2
	400d06f2:	004813        	ssi	f1, a8, 0
	400d06f5:	0121e5        	call8	400d1914 <_ZN5Print7printlnEPKc>
	400d06f8:	fe4fb1        	l32r	a11, 400d0034 <_stext+0x1c>
	400d06fb:	20a220        	or	a10, a2, a2
	400d06fe:	012165        	call8	400d1914 <_ZN5Print7printlnEPKc>
	400d0701:	ffff06        	j	400d0701 <_Z4loopv+0x71>

# Version 2 (slight code change, doesn't crash)

Relevant Code:

	void loop() {
	    Serial.println("Constrain something");

	    // Remove this block though, and it all works again...
	    {
	      static float zzza = 5;
	      static float zzzb = 5;
	      zzza = constrain(zzzb, 1, 9); // <<-- used the other var
	      zzzb = constrain(zzza, 1, 9); // <<-- used the other var
	    }
	    
	    Serial.println("Constraint complete");
	    Serial.println("Halting");
	    while(1) {}
	}

Disassembly of `loop()`:

	400d068c <_Z4loopv>:
	400d068c:	004136        	entry	a1, 32
	400d068f:	fe6521        	l32r	a2, 400d0024 <_stext+0xc>
	400d0692:	fe63b1        	l32r	a11, 400d0020 <_stext+0x8>
	400d0695:	02ad      	mov.n	a10, a2
	400d0697:	012665        	call8	400d18fc <_ZN5Print7printlnEPKc>
	400d069a:	fe6381        	l32r	a8, 400d0028 <_stext+0x10>
	400d069d:	fe5e31        	l32r	a3, 400d0018 <_stext>
	400d06a0:	000803        	lsi	f0, a8, 0
	400d06a3:	fa2350        	wfr	f2, a3
	400d06a6:	4b0020        	olt.s	b0, f0, f2
	400d06a9:	371076        	bt	b0, 400d06e4 <_Z4loopv+0x58>
	400d06ac:	fe5c31        	l32r	a3, 400d001c <_stext+0x4>
	400d06af:	fa1350        	wfr	f1, a3
	400d06b2:	4b0100        	olt.s	b0, f1, f0
	400d06b5:	111076        	bt	b0, 400d06ca <_Z4loopv+0x3e>
	400d06b8:	4b0020        	olt.s	b0, f0, f2
	400d06bb:	251076        	bt	b0, 400d06e4 <_Z4loopv+0x58>
	400d06be:	4b0100        	olt.s	b0, f1, f0
	400d06c1:	051076        	bt	b0, 400d06ca <_Z4loopv+0x3e>
	400d06c4:	000146        	j	400d06cd <_Z4loopv+0x41>
	400d06c7:	000000        	ill
	400d06ca:	fa0100        	mov.s	f0, f1
	400d06cd:	fe57b1        	l32r	a11, 400d002c <_stext+0x14>
	400d06d0:	02ad      	mov.n	a10, a2
	400d06d2:	004803        	ssi	f0, a8, 0
	400d06d5:	012265        	call8	400d18fc <_ZN5Print7printlnEPKc>
	400d06d8:	fe56b1        	l32r	a11, 400d0030 <_stext+0x18>
	400d06db:	02ad      	mov.n	a10, a2
	400d06dd:	0121e5        	call8	400d18fc <_ZN5Print7printlnEPKc>
	400d06e0:	ffff06        	j	400d06e0 <_Z4loopv+0x54>
	400d06e3:	020000        	andb	b0, b0, b0
	400d06e6:	86fa      	add.n	a8, a6, a15
	400d06e8:	fff8      	l32i.n	a15, a15, 60

That said, most of the disassembly is different. A lot is due to small shifts in the addresses of things, as would be expected. But as we see above, a small change has a surprisingly big effect, though that's the nature of compilers and optimizers... Regardless, somewhere in all of this something is going very wrong and I'd really like to know what it is.

marcmerlin, 2017-03-25:
Thanks, 2 thoughts:
1) you said you got get it to crash with compiler optimizations turned off? If so, try that
2) by the same vein, as per my previous comment, replace with a = a +1 which still crashes, and then comment only one of the 2 lines, which is enough to affect the crash
the two above should give you s much smaller assembly diff hopefully

[24 more replies]

closed 2017-04-04 by MartyMacGyver