MartyMacGyver/ESP32-Digital-RGB-LED-Drivers #7: Using this driver seems to shift analogread by one bit to the left (x2)

marcmerlin, 2017-03-26:
I have 
#define JOYSTICK_X_PIN 39
#define JOYSTICK_Y_PIN 34     
#define BAT_PIN 2

All 3 of those values are returning values that are 4095 when they are at 50% after I enable your driver, but for instance if I move my joystick from center to 0, then I get 4095 to 0. 
If I remove your driver, and go back to adafruit, center is 2048 and then the value goes back from 0 to 4096.

marcmerlin, 2017-03-26:
Ok, did some quick tests, and if I comment out these same 3 lines, analogread returns the right values:
  //RMT.conf_ch[RMTCHANNEL].conf1.mem_rd_rst = 1;
  //RMT.conf_ch[RMTCHANNEL].conf1.tx_start = 1;
  //xSemaphoreTake(ws2812_sem, portMAX_DELAY);
As soon as I uncomment them, analogread gets messed up.

Sorry :-/

MartyMacGyver, 2017-03-26:
I'm going to guess this is related to issue #4. Does this issue appear if you test AnalogRead alone (as part of a demo project or such)?

marcmerlin, 2017-03-26:
Sigh, I didn't even check that, and sure enough if I take your basic demo code, display a bitmap on my LCD (i.e. use I2C) and do analogread before and after, it's good.
So, the additional code in my bigger demo now also gets affected on analogread in that new weird way, when I2C isn't crashing as a result.

MartyMacGyver, 2017-03-26:
I'll keep looking into this. Either something is wrong with my code or RMT and its interrupt do not play nice with anything else. I'm hoping Espressif can weigh in on the original #4 thread to clarify if there's something obvious going on here from the platform perspective, or if perhaps there's a limitation of some sort in terms of interrupts.

marcmerlin, 2017-03-26:
I agree that something is definitely wrong here, and I don't think it's your code.
I mean how would a bug in your code change the resolution of analogRead? very unlikely...
Are you in touch with folks from expressif about this? I like the platform so far, but if there are platform/chip/underlying software layer bugs that are causing WTF behaviour like this, it's going to become a lot less desirable to work on this chip, I've already lost many days to bugs that weren't in my code (and likely not yours either), and at this point you've also spent a sizeable chunk of time on this :-/

MartyMacGyver, 2017-03-26:
The only guess I have is that there's a memory leak or pointer/stack corruption somewhere. I still can't rule out my code, but given the other problems I've had with the ESP32 I wonder if it's some bug in the platform code itself, or worse yet in the silicon. I don't like that it all feels very _fragile_...

marcmerlin, 2017-03-26:
@MartyMacGyver agreed. Is there someone you can contact to look into this more deeply considering that we've both done more than due diligence in making sure it's not our fault?

MartyMacGyver, 2017-03-26:
I added a note on the main thread - they are my only points of contact (well, short of the original dev, but I'm sure he'd be surprised as well - I'll ping him anyway.)

MartyMacGyver, 2017-04-04:
As this was resolved via https://github.com/espressif/arduino-esp32/issues/290 I'm closing the issue here.

closed 2017-04-04 by MartyMacGyver