MartyMacGyver/ESP32-Digital-RGB-LED-Drivers #9: Doesn't work with current IDF

paulo-raca, 2017-08-21:
With current versions of IDF, there isn't any output on the pin.

I've identified the problem to be on https://github.com/espressif/esp-idf/commit/17747b33586075e46cadfe0c096dabb7f1a80df2 -- Reverting it makes the leds blink again :)

I would love to submit a PR, but I really haven't figured out those IO matrixes yet, so I'm hoping you can work a proper fix.

Thank you!

MartyMacGyver, 2017-08-21:
That's weird - I haven't used it in a while but odds are it's something in my code. I assume it was the IDF driver you were testing - the Arduino one probably has the same issue.

I'll give it a try when time permits and see if I can see what's going on.

paulo-raca, 2017-08-21:
Neither the IDF or the Arduino versions work out of the box.

I managed to get the IDF version working by reverting that commit.

MartyMacGyver, 2017-08-21:
Well, hopefully it's fixable. I will look at it as time permits, though the next week or so it going to be unusually busy here.

MartyMacGyver, 2017-09-01:
Followup: I'm able to delve into the issue now. I'll update you when I (or ESP) figures something out. I'm hoping it'll be a straightforward fix by either myself or in the IDF.

MartyMacGyver, 2017-09-01:
This auto-closed with my commit, but if it's still not working for you let me know

MartyMacGyver, 2017-09-08:
As I haven't heard from you but this seems to have resolved the main problem and worked for others similarly affected I'm going to close this out.

closed 2017-09-08 by MartyMacGyver

paulo-raca, 2017-09-10:
It took me a while, but I have tested it and works perfectly.

Thank you!

MartyMacGyver, 2017-09-10:
I'm glad to hear it! Thanks for your help tracking down that commit!

commit 7ab9840b (2017-09-01): Updated code to use rmt_set_pin() to work with latest ESP-IDF and Arduino-ESP32

Closes #9