MartyMacGyver/ESP32-Digital-RGB-LED-Drivers #10: Regular guru meditation

cranphin, 2017-09-03:
Hi!

I'm using the (unmodified) Arduino version of ws2812.cpp/.h on PlatformIO
(May 22, 2017 / b1305547a91b6b2ac535cb75f9096114189e0a4e version)

I seem to often get this trace:
```
Guru Meditation Error of type LoadProhibited occurred on core  1. Exception was unhandled.
Register dump:
PC      : 0x40086493  PS      : 0x00060033  A0      : 0x80084fb3  A1      : 0x3ffc0b70
A2      : 0x3ffc3210  A3      : 0x3ffcd04c  A4      : 0x00060023  A5      : 0x3ffc9d70
A6      : 0x00000003  A7      : 0x00060023  A8      : 0x00000000  A9      : 0x00000000
A10     : 0x3ffd72c8  A11     : 0x00060023  A12     : 0x00060023  A13     : 0x3ffc9d40
A14     : 0x3ffc10f8  A15     : 0x00060023  SAR     : 0x00000020  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000008  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000

Backtrace: 0x40086493:0x3ffc0b70 0x40084fb3:0x3ffc0b90 0x400846cc:0x3ffc0bb0 0x400df169:0x3ffc0bd0 0x40081cc9:0x3ffc0c00

CPU halted.
```

Which resolves to:
```
0x40086493 /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./list.c:130
0x40084fb3 /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./tasks.c:4420
0x400846cc /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./queue.c:1992
0x400df169 C:\Storage\PlatformIO\Blink32/lib\ws2812/ws2812.cpp:197
0x40081cc9 xtensa_vectors.o:?
```
The line in ws2812.cpp is [here](https://github.com/MartyMacGyver/ESP32-Digital-RGB-LED-Drivers/blob/b1305547a91b6b2ac535cb75f9096114189e0a4e/arduino-esp32/demo1/ws2812.cpp#L197).

This is with the espressif esp32 arduino version: [b34d18f5769b1dca526fc52926c51abbded593cd](https://github.com/espressif/arduino-esp32/tree/b34d18f5769b1dca526fc52926c51abbded593cd)
(Known in PlatformIO as framework-arduinoespressif32 1.1.5)

Any thoughts on what might be going on here?
The code is running together with a basic RC receiver (interrupt based), wifi (mqtt), and also uses ledcWrite from the esp32 arduino code.

I will be narrowing it down more as I have time, but you might have a aha moment that could save me days of doing that ;)

MartyMacGyver, 2017-09-03:
Before attempting to debug this, you're seeing this on an older version of my code with an older version of the Arduino-ESP32 platform right? Are you also seeing this with the *current* version of the code here and the current Arduino-ESP32 platform as well? Update both and let me know if that helps at all.

If this still a problem with the current library and current Arduino-ESP32 platform, then I wonder if there's an interrupt conflict (I'm not sure how well ledc and RMT work together as it is). At that point I need a way to reproduce the issue.

cranphin, 2017-09-04:
You're correct :)
The reason is PlatformIO has it's own 'stable' versions, which don't update as often, 1.5.1 is the most recent and is identical to  b34d18f5769b1dca526fc52926c51abbded593cd :)

It does allow you to use a 'stage' version, which is a more recent, or head version? I'm not sure about that yet :) Last time it failed to compile, but I will try that again, I might be more lucky now :)

Thank you for giving some direction, I shall investigate further ;)

cranphin, 2017-09-04:
Well, got everything running on new code, first had to figure this out: https://github.com/platformio/platform-espressif32/issues/33#issuecomment-327014178 :)

Haven't seen the issue yet, but now something's funny with my RC receiver code, so have to look into that first ;)

MartyMacGyver, 2017-09-04:
I'm glad to hear it's working better now - hopefully it's all stable once you have the receiver code working! Let me know.

cranphin, 2017-09-08:
Well, so far it looks stable now, so I'll be closing it again, thanks for the help :)

closed 2017-09-08 by cranphin