MartyMacGyver/ESP32-Digital-RGB-LED-Drivers #12: compile error with Arduino

Cubyte, 2017-09-13:
just hoping anyone can point out what i doing wrong or failed to install.
my issue is i get an error in 2 lines of code in the ws2812.cpp file

DPORT_SET_PERI_REG_MASK(DPORT_PERIP_CLK_EN_REG, DPORT_RMT_CLK_EN);
DPORT_CLEAR_PERI_REG_MASK(DPORT_PERIP_RST_EN_REG, DPORT_RMT_RST);

if i // the code it will compile but won't run

MartyMacGyver, 2017-09-13:
Are you fully up to date on your Arduino-ESP32 installation? Which commit are you based on?

MartyMacGyver, 2017-09-23:
If you managed to resolve this yourself let me know.

Cubyte, 2017-09-24:
thanks I updated the arduino ide and esp ide which sorted out the problem .
now it works great thanks for your hard work and great support

closed 2017-09-24 by Cubyte