MartyMacGyver/ESP32-Digital-RGB-LED-Drivers #13: Add a second WS2812B strip

RoelHermus, 2017-09-16:
This software works perfect, but is it possible to add a second WS2812 strip on it?

MartyMacGyver, 2017-09-16:
In theory it should work, but I haven't delved into it yet. As long as one selects a different RMT channel along with a different pin, it should work, but I think the interrupt handling might be more complex - either way, I'll take this as a feature request and start looking into it!

RoelHermus, 2017-09-16:
Thank you so far. I'm studying a couple of hours on it and it's a eye opener for me how this works. But it's not easy (for me) to change it for 2 channels. Your point about the interrupt is clear, but not simple to fix it for me. A fast work around is add the 2 strips together.

JaykeMeijer, 2017-09-16:
Very coincidentally, I have been working on this this evening. I've got it to work but need to fix up some stuff. Will update as soon as I have done so.

RoelHermus, 2017-09-16:
I like to be your first tester!

JaykeMeijer, 2017-09-16:
I will share it tomorrow, just want to have a fresh look at it in the morning to make sure there are no obvious mistakes (been staring at it for about 7 hours now).

@MartyMacGyver shall I create a branch in this repo or would you prefer that I share the modified source files?

MartyMacGyver, 2017-09-16:
Just fork my repo and we can pull it in via a pull request in the end. I'm interested in seeing what you come up with - I'm pondering the API now.

MartyMacGyver, 2017-09-17:
I'm looking at the code and wondering if this wouldn't be a lot easier as a class. That would make this C++ only though, but it would be much easier to instantiate multiple instances.

MartyMacGyver, 2017-09-17:
I've set up a testbed and am looking at the code. If anything will be tricky it will be the interrupt handler.

For simplicity I'll be working on the Arduino version - once that's stable then porting to native ESP-IDF should be trivial (they're normally identical).

I'll make changes on the [feature-multistrand](/MartyMacGyver/ESP32-Digital-RGB-LED-Drivers/tree/feature-multistrand) branch until this is ready to be merged.

The new API will either require routinely referencing the RMT channel or an object equivalent - making this a class might help in the end, but as the interrupts need to cooperate I'm not sure how much that would actually compartmentalize things. I'll extend the syntax of what I have now to see how the interrupts work out before making such a change.

I appreciate suggestions and pull requests - I may or may not merge a given request as-is, but such working ideas will definitely help guide the design of this!

JaykeMeijer, 2017-09-17:
My changes can be found on https://github.com/JaykeMeijer/ESP32-Digital-RGB-LED-Drivers, https://github.com/JaykeMeijer/ESP32-Digital-RGB-LED-Drivers/commit/25e7e58d235317d025526f5012716708b18d7695. I have done some minimal testing with 2 strips, but I think it should work with up to 8. The changes have been made to the Arduino version only.

The API is changed in the following way:
- Initialization: `ws2812_init(int gpioNum[], int ledType, uint8_t stripcount)`. So the instead of passing a single gpio port, an array of desired ports is provided. The strips will be assigned in the provided order. Additionally, the number of strips needs to be provided.
- Setting: `ws2812_setColors(uint8_t strip, uint16_t length, rgbVal *array)`. So the strip to set is provided too. The strip number starts at 0 and follows the order as provided for the gpio pins during the initialization.

I did a 'minimal effort' implementation, as I haven't really done any embedded programming before and I haven't done any C/C++ in a few years. Basically, all I did was turn the global variables into arrays. As a result, there are some caveats:
- Only successive RMT channels can be used. The first gpio pin provided is assigned to channel RMT_CHANNEL, the second to RMT_CHANNEL + 1 etc. Changing this should be easy enough, but requires a bit more administration. Should be easy in a class based approach though.
- Only strips of the same length can be used. This should be easily fixed by making `ws2812_len` an array too. It was simply not required for my purpose.
- I realized this morning that it's C++ only at the moment, as I use `new` rather than `malloc`. Again, should be an easy fix.

As you've said, the RMT interrupts are the tricky bit. The bit that took me a while to figure out is that the channels aren't array/index based, but 'hardcoded' (eg `RMT.int_ena.ch0_tx_thr_event`, note the `ch0`), hence the rather large switch/if-elseif blocks in my changes (on a positive note - this does allow using any RMT channel, I think before only 0 could be used?).

As said, this is a minimal effort implementation. I think that changing it to be class based is definitely the way to go. The interrupt handler would have to stay outside though, and somehow trigger the functions in the class. If you want to keep it C compatible, maybe store all the variables in a struct and have the user provide that to the setColors function?

MartyMacGyver, 2017-09-17:
Thanks, I just saw your reply!

I've been doing a fair amount of work on this tonight as well - I will dig into this more this evening and look at your code as well. I've uploaded my changes so far to the [feature branch](/MartyMacGyver/ESP32-Digital-RGB-LED-Drivers/tree/feature-multistrand).

The switches may or may not be unavoidable (I think there is some indexing trickery one could do, but that optimization can wait for now.)

The interrupt handler is of the form `void ws2812_handleInterrupt(void *arg)`

The interrupt handler is allocated via `esp_intr_alloc(....., ws2812_handleInterrupt, void * arg, .....)`

I'm trying to figure out how to make use of `arg` - if you run across anything about that, let me know. It might be useful (and if nothing else I'd like to know why it doesn't want to work!)

Debugging is working again - setting `DEBUG_WS2812_DRIVER 1` in the header does it. However, the LEDs will look like hell with that enabled - its more for debugging variables and such than realtime debugging.

JaykeMeijer, 2017-09-17:
The problem with the interrupt handler is that I don't think the arguments can be used to tell you which channel caused it. It's simply called if there is an event in the entire RMT. Note that `
  esp_intr_alloc(ETS_RMT_INTR_SOURCE, 0, ws2812_handleInterrupt, NULL, &rmt_intr_handle);` does not have any way to assign this to a single channel. You can provide arguments but these will be identical regardless of the channel that caused them.

I solved this by checking the events for all channels (repeating the block you had in the handler 8 times, once for each channel (replacing `ch0` with `ch1`, `ch2` etc), and calling `copyToRmtBlock_half` with the channel number as argument. Inside `copyToRmtBlock_half` I then use the channelnumber to select the correct variables from the arrays. This works well for me, but I am slightly worried about the large number of compare statements handled for each event. I only have 2 strands at the moment, so I don't know how it will behave when you are trying to handle up to 8 at a time.

At first I thought you don't actually need to store the variables (all the `ws2812_` vars) per channel, but I think due to the interrupt structure it might be handling multiple channels at the same time. I now store those variables as array, but the ideal solution would probably have an object or struct per channel, storing `ws2812_buffer`, `ws2812_pos`, `ws2812_half` and `ws2812_bufIsDirty` (and `ws2812_len` if you want to support strands of different length) inside that.

MartyMacGyver, 2017-09-17:
You're right! I think the '0' param threw me off (at a glance I thought it was the RMT channel - hadn't dug into it yet).

Even if it can't be used for what I first imagined I still wonder, academically, what "arg" is and/or what my syntax error is with that (since I've yet to get back from it what I put into it). 

We both took similar approaches with the API - I ended up just making it an array of structs which helps a lot with encapsulation.

I see how you handled the interrupts and I see your point. I have room to test 3-4 channels so I can give this some solid exercise once I get the code in place tonight. And yes, I'll consolidate and clean up those buffer vars too.

RoelHermus, 2017-09-17:
I did a simple test with the changes made by Jayke with two short strips.  In the demo I change pixel array in pixel1 and pixel 2 array. Fill them with different data and let run the program. Both of the stips working fine with there own data. It looks good! Thanks so far.

JaykeMeijer, 2017-09-17:
> Even if it can't be used for what I first imagined I still wonder, academically, what "arg" is and/or what my syntax error is with that (since I've yet to get back from it what I put into it).

I assume it's there so you can maybe use one handler for multiple type of events? Or maybe if you want to use less global vars?

Regarding why it doesn't work for you, I think it's because you're pointing to a variable created in the scope of the `init` function. That will be gone once the init function is done. For me, passing on the address of a variable defined globally it seems to work, as well as passing on a variable `malloc`'ed inside the init.

@RoelHermus Great to hear! I've been using it to further work on my project today and it has worked without problems. Next weekend I'm planning to cut my strands up into 4 pieces in total, so I'll be able to see how it works with more than two.

JaykeMeijer, 2017-09-17:
By the way, I just found the source of the RMT struct (http://esp32.info/docs/esp_idf/html/da/d5e/rmt__struct_8h_source.html) and saw the following definition:
```
uint32_t status_ch[8];
```

so in your code that would be accessible as `RMT.status_ch[<channel]`. Perhaps that can be used to more efficiently process the interrupt. I don't have time to try it out now, but I thought I'd mention it.

MartyMacGyver, 2017-09-17:
It turns out I was passing in `(void *)&blahTODO`, but I was dereferencing it to an address. Making that var static and/or changing the scope ultimately helps too (though it happened to work here once I got the derefs fixed, it was lucky at best given the scope change). There are a couple of ways to pass vars, but this was neither.

Pointers are fun... Now I can rip that bit of code out - it was educational!

I should have a lot more code and data to talk about by tomorrow.

MartyMacGyver, 2017-09-17:
Interesting - will look at that status array and give it some thought too!

MartyMacGyver, 2017-09-18:
FYI the latest code for the ESP32's RMT and its struct are here (the one you mentioned may be out of date):

https://github.com/espressif/esp-idf/blob/master/components/soc/esp32/include/soc/rmt_struct.h
https://github.com/espressif/esp-idf//blob/master/components/driver/rmt.c

The latest docs for RMT are here:

http://esp-idf.readthedocs.io/en/latest/api-reference/peripherals/rmt.html

And the latest docs for interrupt allocation are found here:

http://esp-idf.readthedocs.io/en/latest/api-reference/system/intr_alloc.html

Looking at the source, that `status_ch[]` is exposed as `rmt_get_status()`, which doesn't seem to do what one might hope WRT interrupts. Which is fine... I've got a few ideas on how to poll the interrupts anyway.

That all said, I wonder if you'd ever see a situation where two interrupts get processed at the same time... or at least whether you'd ever want that.

In the present `ws2812_setColors()` we fill the buffer, fill half the RMT block with it, and if the buffer was longer than the full block we create the semaphore, kick off the transmission, and let the interrupt handle subsequent half-blocks til its done. Then we block on the semaphore til we get it back from `ws2812_handleInterrupt()` (which only happens when the last block has been transmitted).

I see how you handled the semaphores (one per strand, effectively). The question I have is, what happens if an RMT interrupt occurs while `ws2812_handleInterrupt()` is handling another RMT interrupt?
 Or will that even happen? I'm concerned that this might adversely affect timing, but I suppose we shall soon see.

MartyMacGyver, 2017-09-18:
Also, I did end up doing some things to hopefully make this work better than case/switch or if/else when dealing with multiple channels. I noted the format in the code, but to address your concern I don't think being clever will be so risky, as the RMT bit mappings are to hardware registers unlikely to move around when it comes to the ESP32.

MartyMacGyver, 2017-09-18:
Latest code is on the [feature branch](/MartyMacGyver/ESP32-Digital-RGB-LED-Drivers/tree/feature-multistrand). Below is a short video of this running 4 strands, each off a different RMT channel.

I'll be working on a better demo - this shows them being run sequentially but I'd like to show interleaving (it can't be precisely simultaneous of course, but it should look virtually the same).

https://i.imgur.com/y9h3V55.gifv

Thank you once again for your help with all this!

RoelHermus, 2017-09-18:
@MartyMacGyver  I download your update and try it in a ESP32. The result is fantastic. It works great. Today I will try it in my project. Thank you very much. Great job!

RoelHermus, 2017-09-18:
oeps wrong button

MartyMacGyver, 2017-09-18:
Thanks! Still working on it before it ends up on master so watch this space.

JaykeMeijer, 2017-09-19:
Just had a go with it, seems to work fine! Will play with some more strands coming weekend probably.

MartyMacGyver, 2017-09-19:
Thanks!

However, either there's a subtle bug in it or C++ on the ESP32 is really peculiar - see the current version of the demo (which is set up to test three strands right now). If I use `Rainbower` and instantiate that class with `static` and without using `new` it all seems to work fine, but if I omit `static` or if I use `new` and point to it, strange things happen (it doesn't crash, but the pixel data ends up being unexpected). One instance? Seems OK. More than one instance? The second instance gets weird data.

I looked at that for a long while last night without progress. I don't think there's a memory leak but I'm not sure. Might be related to the silicon bug `When the CPU accesses external SRAM through cache, sometimes random read and write errors occur.` ([see the errata for details](https://espressif.com/sites/default/files/documentation/eco_and_workarounds_for_bugs_in_esp32_en.pdf)).

I'll be investigating that separately.

MartyMacGyver, 2017-09-20:
Another peculiarity - even when all the `delay()` calls are zero, the Arduino version of this current demo runs visibly much faster during the drawing loops than the ESP-IDF compilation of basically the same thing. I have no idea why...

commit e0c0e01a (2017-10-04): Everything changed... and improved greatly (summary below)

Refine the API quite a lot - much was renamed and improved
Cleaned up TODOs and cruft; tightened up code
Updated documentation and .gitignore

LED sub-versioning added
Works with RGBW strands as well (e.g., WS2813)

Lib is C++ but now works properly when called from C as well
Add wrappers in header for greater C compatibility

Added sdkconfig.defaults with documentation in comments
(ESP-IDF default speeds are lower than desired by IDF default.)

Dr. Subtlebug or: How I Learned to Stop Worrying and Love the C++
memset() comes from "str [... cut]
closed 2017-10-04 by MartyMacGyver