MartyMacGyver/ESP32-Digital-RGB-LED-Drivers #15: How to use only ws2812 library without arduinoish.hpp?

neosarchizo, 2017-09-22:
Hi! Thanks for the great library!

I want to use only ws2812 library in my esp32 project without `arduinosh.hpp`. My project has the following folder structure:

```
/ (root)
/main
/components
/components/ws2812
/components/ws2812/include
```

My main source codes are like following codes.
```c++
#include <esp_log.h>
#include <freertos/FreeRTOS.h>
#include <freertos/task.h>
#include <math.h>
#include "ws2812.h"

#include "sdkconfig.h"

static char tag[] = "neosarchizo";

const int DATA_PIN = 17;
const uint16_t NUM_PIXELS = 3;
uint8_t MAX_COLOR_VAL = 32;

rgbVal *pixels;

void app_main() {
	ESP_LOGD(tag, "start");
	ws2812_init(DATA_PIN, LED_SK6812);
	pixels = (rgbVal*)malloc(sizeof(rgbVal) * NUM_PIXELS);

	while(1) {
		for(uint16_t i=0; i<NUM_PIXELS; i++) {
			pixels[i] = makeRGBVal(MAX_COLOR_VAL, 0, 0);
		}
		ws2812_setColors(NUM_PIXELS, pixels);
		vTaskDelay(1000/portTICK_PERIOD_MS);
		for (uint16_t i=0; i<NUM_PIXELS; i++) {
			pixels[i] = makeRGBVal(0, MAX_COLOR_VAL, 0);
		}
		ws2812_setColors(NUM_PIXELS, pixels);
		vTaskDelay(1000/portTICK_PERIOD_MS);
		for (uint16_t i=0; i<NUM_PIXELS; i++) {
			pixels[i] = makeRGBVal(0, 0, MAX_COLOR_VAL);
		}
		ws2812_setColors(NUM_PIXELS, pixels);
		vTaskDelay(1000/portTICK_PERIOD_MS);
	}

	vTaskDelete(NULL);
} // task_hmc5883l

```

But I get following errors when I build it.

```
/Users/neosarchizo/esp/aqk-device/build/main/libmain.a(aqk_device_main.o):(.literal.app_main+0x10): undefined reference to `ws2812_init'
/Users/neosarchizo/esp/aqk-device/build/main/libmain.a(aqk_device_main.o):(.literal.app_main+0x14): undefined reference to `ws2812_setColors'
/Users/neosarchizo/esp/aqk-device/build/main/libmain.a(aqk_device_main.o): In function `app_main':
/Users/neosarchizo/esp/aqk-device/main/./aqk_device_main.c:19: undefined reference to `ws2812_init'
/Users/neosarchizo/esp/aqk-device/main/./aqk_device_main.c:26: undefined reference to `ws2812_setColors'
/Users/neosarchizo/esp/aqk-device/main/./aqk_device_main.c:28: undefined reference to `ws2812_setColors'
/Users/neosarchizo/esp/aqk-device/main/./aqk_device_main.c:33: undefined reference to `ws2812_setColors'
collect2: error: ld returned 1 exit status
make: *** [/Users/neosarchizo/esp/aqk-device/build/aqk-device.elf] Error 1
```

What did I do wrong?

MartyMacGyver, 2017-09-22:
A lot will change in the next release (big feature in progress). That said, if you clone the current version on master, you should be able to build the esp-idf demo. Use that as a guide.

Also, this may be a C thing... my demos are in C++ (as is the library) and I've not looked into how they bind with a main program in C. It may be that this is the problem.

`arduinosh.hpp` just has some helpful glue code for the (C++) demo... you can instead copy that code in directly. However, again this C vs C++ thing may be happening here.

If after examining the latest version and verifying your structure (and component makefiles!) matches, then perhaps attaching a tarball with JUST the source and makefiles for your demo would help. `make clean` first and you can skip the sdkconfig file(s) too.

I cannot guarantee this will ever work with regular C, but its worth figuring out where this problem is.

neosarchizo, 2017-09-23:
For this, I create pull-request.
I added demo2 for projects that work without `arduinoish.hpp`.

MartyMacGyver, 2017-09-23:
Thank you! I will look at it tonight and integrate it (I may make changes now or later but I'll keep it as a C demo).

MartyMacGyver, 2017-09-24:
Not closed yet...

neosarchizo, 2017-09-24:
Oh sorry. If you need any help about maintaining it, just let me know.

MartyMacGyver, 2017-09-24:
You're the second person to ask about it, so it's definitely on my todo list.

commit e0c0e01a (2017-10-04): Everything changed... and improved greatly (summary below)

Refine the API quite a lot - much was renamed and improved
Cleaned up TODOs and cruft; tightened up code
Updated documentation and .gitignore

LED sub-versioning added
Works with RGBW strands as well (e.g., WS2813)

Lib is C++ but now works properly when called from C as well
Add wrappers in header for greater C compatibility

Added sdkconfig.defaults with documentation in comments
(ESP-IDF default speeds are lower than desired by IDF default.)

Dr. Subtlebug or: How I Learned to Stop Worrying and Love the C++
memset() comes from "str [... cut]
closed 2017-10-04 by MartyMacGyver