MartyMacGyver/ESP32-Digital-RGB-LED-Drivers #17: esp-idf demo sets LEDs to full white

mattcollins107, 2017-10-28:
I'm running an ESP32 Thing from Sparkfun, driving a Neopixel 12 LED ring.  

The Arduino demo works without issue after configuring the "strand" to be 12 pixels long.  The esp-idf demos, though, both set all LEDs to full white brightness.  Have tried a few times to make sure I didn't screw anything up, and either I'm always screwing something up or there's something incompatible somewhere.  Any ideas?

MartyMacGyver, 2017-10-29:
If you're using the very latest version of these drivers as well as the very latest IDF this ought to be working... however, I'm not as familiar with the ESP32 Thing so maybe it's a timing issue?

There are certain timing defaults set in `sdkconfig.defaults` that *might* be an issue unique to that board, but I don't think that's the problem here. I do recall there being some unique things about the Thing versus the regular board so that might be part of the problem. See this thread: https://esp32.com/viewtopic.php?t=2800

So, per that thread, run `make menuconfig` and set **Component Config -> ESP32-Specific -> Main XTAL frequency** to **26 MHz** and try it again (I'm pretty sure my extra defaults won't affect this but let's see how this one change goes first).

mattcollins107, 2017-10-29:
That fixed it entirely.  Great find and thanks for the help!

closed 2017-10-29 by mattcollins107

MartyMacGyver, 2017-10-29:
You're welcome - Glad it worked out!