MartyMacGyver/ESP32-Digital-RGB-LED-Drivers #18: Not working in latest version of arduino-esp32

dsteinman, 2017-11-03:
Hi, I was successfully using your library, and then (for other reasons) I upgraded to the latest version of arduino-esp32 at https://github.com/espressif/arduino-esp32 and your library stopped working.  Have you tried upgrading recently?

It's unclear to me why it's not working, the console isn't showing any error messages, the pixels just don't light up anymore like they used to.  I was using the older version of your code from February, and upgraded to your latest, and tried each setting LED_WS2812B_V1 to V3, and some different pins.  No luck, I'd appreciate your help.

MartyMacGyver, 2017-11-04:
Interesting - I'll take a look at it shortly.

MartyMacGyver, 2017-11-04:
I'm not sure what the problem might be but a lot changed in the latest release. That said, I just rebuilt and re-flashed demo1 and it's working fine for me based on the latest Arduino-ESP32 (commit 126674c) and Arduino 1.8.5 IDE.

dsteinman, 2017-11-04:
Hmm, I suppose it could be due something else, I am using wifi & bt & neopixels at the same time btw.  I tried out the Neopixel Bus library and it did work so I was able to get going again.  I'm not sure if their methods of interacting with the neopixels differs from yours.

MartyMacGyver, 2017-11-04:
Did you try the demo alone, perhaps stripped down to one strand (just to see it work)?

dsteinman, 2017-11-04:
Yup, I tried everything I could think of.  I guess you can close this issue unless it crops up again for other people.

MartyMacGyver, 2017-11-04:
Hmm. I'd rather keep this open til you're back up and running there.

Can you describe your setup (for the purpose of testing against just one strand)? And are you able to drive the LEDs with anything else right now, just to rule out electrical problems, broken wires, and such?

dsteinman, 2017-11-04:
Okay, I have double checked, yes you close this ticket.  I was able to get your "demo1" code working now.  I'm not sure where my error was, but it could have been due to some unrelated code, pin #, or library changes I was making.  Sorry for the false alarm!

Here is my gist for the working single strand (in my case 24 pixel ring on pin 15) https://gist.github.com/jaxcore/73593b3a2c13e3757186f8f02d31c690

MartyMacGyver, 2017-11-04:
I'm glad to hear it's working now!

closed 2017-11-04 by MartyMacGyver