MartyMacGyver/ESP32-Digital-RGB-LED-Drivers #19: What are the default Arduino Values for Tickrate and CPU needed?

xMAC94x, 2017-11-15:
Hi, i have a question, what are the needed values i need to change via ` make menuconfig` for CPU Frequency and Tick rate?
If i leave the values as they are. Should the LEDs still work or not work at all ? because i have problems getting my WS2812B to work (from this site: https://www.banggood.com/100Pcs-DC-5V-3MM-x-10MM-WS2812B-SMD-LED-Board-Built-in-IC-WS2812-p-979463.html?rmmds=detail-left-hotproducts&cur_warehouse=CN ).
I have successfully tested the RMT interface via an integrated LED (set RMT to pulse reaaaaaally slow like just 4 times a second). But i didn't get my WS2812B to work. Any tips here ?

MartyMacGyver, 2017-11-15:
Those settings are in `sdkconfig.defaults`, however, even at a slower system clock rate the only effect is slower overall performance - it shouldn't affect the timing of the pulse trains coming out of the RMT peripheral (it'd be pretty clear if it did).

It may not be the driver but the devices themselves, or some other part of the setup (they might not be WS2812 but some variant like the SK6812). I have no way of testing the devices you have, but I'd start with selecting different variants from the library. The likely variants of the WS2812 include:

```
  LED_WS2812_V1
  LED_WS2812B_V1
  LED_WS2812B_V2
  LED_WS2812B_V3
  LED_SK6812_V1
```

Failing that (which is a pretty quick test), I'd also give them a try with some other library, just to ensure they work in the first place.

One last thing: ensure the LEDs are getting 5V logic data. A 3.3V logic level signal can drive them *sometimes* (as 3.3V is right about the lower edge of 'high' in 5V logic), but I do level shift the signal to ensure it works reliably. A 74HCT125 works nicely but tie the unused inputs low or you'll get noise/crosstalk.

At the recent Espressif event in SF I learned about a neat hack to accomplish this: power the first RGB LED at 5V *but through a forward-biased diode with a fair amount of voltage drop* - the diode drops that LED's input voltage to around 4.3 V (the rest of the LEDS are powered at the full 5V). That first undervolted LED should still work (perhaps a bit dimmer than the rest), but as its operating voltage is lower, its threshold for detecting logic high on input is proportionally lower too, making it more tonerant of 3.3V logic inputs. The output of that first LED (which is a *regenerated* pass-through signal after that LED absorbs its 24 bits of data) is high enough (being referenced to about 4.3V) that it can drive the next LED (running at the normal 5V) with reasonable logic thresholds. The rest of the LEDs get 5V logic signals.

I hope that helps!

xMAC94x, 2017-12-03:
Wow, that is alot information, thank you.
In the past 2 weeks  i tried a lot of programs and even tried setting up the esp-idf in windows. However nothing helped. i could get random colors from time to time but it didn't worked. Yesterday i tried some other LEDs of the same type but from a different packange. And within a few minutes i got the blue led working reliably. LED_WS2813_V2 seems to be working best. After a bit of tweeking here and there it got all colors reliable working. Thats great. My setup is having the LEDS at 5V with 100 Ohm reducing the signal to 4.7 V. The Datasignal is 3.3 V with a 470Ohm resitor. Even throug you mentioned that 3.3 V is quite low for the LED it seems to work. However to be sure i also bought some 74HCT125 as you mentioned just in case :)
So in the end it seemed to be some defect LEDs or me destroying some because of my bad solder skills. Again thanks alot

closed 2017-12-03 by xMAC94x