MartyMacGyver/ESP32-Digital-RGB-LED-Drivers #20: Running at same time as i2s

srlevitt, 2018-02-27:
[Great library, like it lots] I am trying to continually update a 64-pixel strand whilst playing a wav file through i2s. Occasionally, the strand seems corrupted with odd pixels being lit. The led display and the wav playing are in different FreeRTOS tasks. Is it possible the i2s output may be corrupting the RMT buffers?

MartyMacGyver, 2018-03-03:
I don't have an i2s device to test this use case against - but out of curiosity, which specific i2c device are you using? I've never tried this use case.

The specific revision of ESP32 and carrier board, the exact LED type, and that the strand has a solid 5V supply to it (not just USB) are also potential factors here. And of course your code and the version of the IDF you're using also have a potential impact.

Short of one peripheral corrupting the other, one possibility is that interrupts are conflicting - if the RMT peripheral and the i2s peripheral are interrupting each other, the display output (which is very sensitive to timing) may be impacted.

It's also possible that the timings for your LEDs are a little off - how often does the problem occur? Does it vary with the input?

srlevitt, 2018-03-03:
I've done some more digging. I'm thinking now it might be caused by just reading my wav file from SPIFFS without even touching i2s. I get similar effects when I reading a bitmap from SPIFFS at the same time as sending new led pixels.

MartyMacGyver, 2018-03-05:
It *could* be one of the more subtle silicon bugs if it's an older device (involving memory reads). Without more information it's impossible to say.

srlevitt, 2018-03-05:
I'm using a Wemos esp32 rev 1 board.

MartyMacGyver, 2018-03-06:
Interesting. The current IDF ought to mitigate any such effects. Beyond that (and without a setup to compare to) it's hard for me to say what might be going on.

fabiuz7, 2018-05-10:
Any news? I'm going to try an i2s device combined with a microsd card (the wav source). I'm pretty sure that the integrated SPIFFS is really slow compared to a microSD. Note that the microSD, rarely, takes a lot of time to retrieve even a small chunk of data (4bytes). Further details will come by the end of the day.

I'm not sure about my code understanding, hence another question: given that the RMT buffer dedicated to a single strad is 64 pulses long, and considering NeoPixel timings, every 86microsecond the RMT buffer has to be refilled, right? Considering you halved this size, every 43microsecond you need to send new data to the buffer. This is seems a VERY tight constrain, I'm wondering if this will kill concurrency while updating the strand...