MartyMacGyver/ESP32-Digital-RGB-LED-Drivers #25: decoding ws281x

ririfonfon, 2018-05-31:


MartyMacGyver, 2018-06-01:
If you're going to simply chain them all together, there should be no need for any intermediate devices to keep the signal going - as long as the strings are adequately powered at regular intervals the data signal will get regenerated at each LED, so it should go indefinitely. However, this design (which you describe) will suffer from lower frame as the number of pixels increases (which you seem to want to avoid).

One way to avoid that is to have all the ESP32s talking to each other and starting each refresh of their respective segments simultaneously. A mesh network would be ideal, but I haven't tried it and I'm not sure how far along Espressif is with it:

http://esp-idf.readthedocs.io/en/latest/api-guides/mesh.html

I hope that gives you some direction!

ririfonfon, 2018-06-01:
indeed a mesh network, will not work:
- carry wifi
- random sync
- the installation will be in the city center.2.4 ghz disrupt
- etc.
for me the ideal would really be to have a lib of ws281x in input

MartyMacGyver, 2018-06-02:
I have a relatively clever idea how this could work at scale and at high speed IF the full set of sequences is known in advance (or is programmatically generated in a repeatable way). Can you tell me more about what the sequence you're displaying is like?

Driving a given strand isn't difficult, decoding that kind of data as an input might be more challenging, but there are some short-cuts one can take if the input is known or fully generable at each ESP32 that maintain sync without sacrificing the frame rate.

ririfonfon, 2018-06-02:
I count on one esp every 20 m. this sp will control around 200 ws 2812 and 3 pwm (rvb). the first sp will be the master. the animation is simple we send white led by led. (ws 1 white then ws 1 black with ws 2 white then ws 1 black ws 2 black ws 3 white ...) when we arrive at ws 201 l esp next pwm rvb full and ws 1 white ...

so the decoding of the ws in will only be on id 201.

MartyMacGyver, 2018-06-02:
Is the complete pattern known in advance? And what is the expected refresh rate for the whole system?

I feel a drawing might help make more sense of this.

If the data is only coming from the first ESP32 (and periodically decoded by subsequent ESP32s along the way) the refresh rate will be terrible. If the pattern is known then one can have each ESP32 doing its part, using the output of the previous strand ONLY for clock sync (there could be a way to use this to also initialize the pattern, but after that the pattern is remembered by each device, greatly increasing the refresh rate).

ririfonfon, 2018-06-02:
yes output is used ws only to synchronize the following esp. each species plays are senario.

ririfonfon, 2018-06-02:
i buy this https://www.aliexpress.com/item/10pcs-Luminosity-Sensor-GY-2561-TSL2561-Light-Module/32365876484.html?spm=a2g0s.9042311.0.0.3d054c4dOKifAW. and put one more ws at the end. detect it with this sensor.

MartyMacGyver, 2018-06-02:
Then why use a sensor? Just use the output of the last LED in a strand as a trigger (NOT as a data input, just a trigger to maintain timing of thpre-defined sequence loaded into each ESP32).

ririfonfon, 2018-06-02:
I try. But it does not work. with a circular test ws of 12 pixels. I generate my frame of 13 pixels and the trig triggers at 3 pixels.