MartyMacGyver/ESP32-Digital-RGB-LED-Drivers #27: LED Drivers for dummies

remcoNL, 2018-08-28:
Hi,
I took the example 2 (for Arduino) and placed it on my esp32. I changed the number of leds and the gpioNum. 

The result is a Guru Meditation Error and 24 red leds flickering. 

Is there some more guidance on how to use this lib properly? 

Thanks!

remcoNL, 2018-08-28:
I found this lib that looks a bit easier to me. [https://github.com/RoboticsBrno/SmartLeds/tree/master/examples/basicExample](url)

However, it throws the same error, so must be something else.

MartyMacGyver, 2018-08-28:
Are you powering the LEDs with a separate power supply? The red flicker suggests the LEDs don't have sufficient power: the typical ESP32 breakout board won't have anywhere near enough power on its own to drive 24 LEDs at anything but the lowest brightness levels, if that.

If power is too low it may in turn cause a brown-out on the device, leading to errors (not sure if that's what the error is about though).

remcoNL, 2018-08-28:
Thanks! It's maybe is not, since i've a separate power supply and the colors are more or less fine with the FastLed lib until i use WiFi, which then indicates it is a power supply problem )-;

Also it might be the hardware or the logic converter.

remcoNL, 2018-08-29:
Finally it was: bad soldering on the second ground pin of the converter, plus calling a non existing pixel, plus certain colors needing too much power (and others are fine). Setting the brightness to 25 "solved" this. I'm so glad this is just a hobby!

MartyMacGyver, 2018-08-29:
Sounds painful but I'm glad you found the root cause!

closed 2018-08-29 by MartyMacGyver