MartyMacGyver/ESP32-Digital-RGB-LED-Drivers #29: Running the RMT channels in parallel

eyer1951, 2018-12-12:
According to the ESP32 data sheet ([here](url
https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/peripherals/rmt.html)) “The RMT has eight channels numbered from zero to seven. Each channel is able to independently transmit or receive data.”
This driver can use up to 8 RMT channels, but as written does not send data in parallel. This is because there is a wait for semaphore, which pauses delivery of the data for each strand until all the data for that strand has been sent.

For the animation, the demos call drawNext() continuously for a period of seconds, but each call to that function must only perform 1) changing pixel color values (rainbow colors, or scanner); and 2) digitalLeds_updatePixels for each strand.  If digitalLeds_updatePixels() would only transfer the pixel data to a buffer, it would run very fast. But the buffer won't be ready to accept more data (set of different pixel colors) until the prior data has been sent, and that can take a few milliseconds. This is where the semaphore comes in. In digitalLeds_updatePixels() there is a call to xSemaphoreTake(pState->sem, portMAX_DELAY) which waits indefinitely for that semaphore, which is sent by the interrupt when the last data has been sent to the strand.

This driver would be greatly improved if it could run the “update pixels” for each strand in parallel (concurrently). It would mean a very large number of pixels could be refreshed at a nice rate (30 or 40 Hz).

I think all that is required is to relocate the xSemaphoreTake() statement (wait for completion), and let other strands start up in parallel. The call to updatePixels would need only to wait at the end for ALL the strands data to be sent (wait for last one to finish).

eyer1951, 2019-01-15:
Update: verified that four RMT channels can be run in parallel, and that the CPU can be doing other things while the strands are updated. I tested with four strands of 200 LEDs each.  My implementation is sort of sloppy (needs a pro to make it clean). I added a "done" flag to digitalLeds_stateData which is cleared initially then set  from within the interrupt when the last pixels are sent. Then checked that flag before digitalLeds_updatePixels is called, to ensure new updates were not starting before the prior update of that strand was done.  Changed 	
  xSemaphoreTake(pState->sem, portMAX_DELAY);
to
  xSemaphoreTake(pState->sem, 0);

MartyMacGyver, 2019-01-16:
I've been swamped lately. Only four channels? What limits that - is it just too much throughput above that point?

MartyMacGyver, 2019-01-31:
I finally had some time tonight to reproduce the issue visually:

[image]

In the coming days I'll dig into this more.

MartyMacGyver, 2019-01-31:
Also, the updating of n strands at once is a special case - the general case is any strand that can be updated should be able to be updated without interrupting the others. Thus, a short strand could sequence normally along with a longer strand, without either waiting for the other.

Is that what you achieved with your changes?

eyer1951, 2019-02-01:
Agree. Excellent point. Any strand should be able to updated any time without interrupting or affecting any other one. I didn't try to achieve that with what i did. I made a loop that waited for all to be done before starting again to refresh all.  My assumption was that all strands were equal length. If you can make them all independent, that would be great!

eyer1951, 2019-04-03:
While it could be helpful to upgrade the performance of this utility, I would now recommend that people check into FastLed [https://github.com/FastLED/FastLED](url). Happily, it supports ESP32 hardware features, including use of the RMT channels in parallel. It works great, and has all sorts of excellent utility routines for pixel manipulation, like easy support for HSV color. And 8-bit optimized math and trig.  

If I had known initially about FastLED, I wouldn't have gone in the direction of working with ESP32 Digital REG LED Drivers. I did learn a lot though.

MartyMacGyver, 2019-04-04:
I've also used [FastLED ](https://github.com/FastLED/FastLED) for other projects, it's definitely handy.

https://github.com/FastLED/FastLED

MartyMacGyver, 2019-07-06:
After some hardware updates, I'm turning my attention back to this problem. It turns out FastLED can't handle something mine can - **RGBW**. It's not clear when they'll support it (they aren't doing major features these days, and handling RGBW in their codebase is very non-trivial).

I've been rewriting the library interfaces, and have started poking at the core code that handles semaphores and so on. I took a look at [how FastLED does it](https://github.com/FastLED/FastLED/blob/bbcbb4017ced2f63e521bc1cf7e7b9669da1b1cb/platforms/esp/32/clockless_rmt_esp32.h), thinking they might have some similar code - turns out, [it's actually a variant of my code](https://github.com/FastLED/FastLED/issues/507#issuecomment-333298905). Open source FTW! :-D

I hooked up my analyzer and did some tests with FastLED to establish a baseline - basically, during a refresh the RMT channels are fully parallel, and the longest strand sets the pace (so if you have one long strand and several short ones, the timings are entirely up to the longest strand). This is acceptable to me, as trying to maximally pack the strands isn't really feasible (at least not without having a lot more complexity). And it handles multiple LED timings, which I wasn't sure would work right... but interestingly, it handles all the WS2812 variants with the same timings (which isn't optimal IMHO - I wonder if they get failure edge cases because of that?)  Anyway, here's their timings:

[image]

I'd already started looking at a global semaphore to replace the individual ones, so I'm on the right track but the interrupts are vexing (they seem to be using the default ISR, and perhaps I should as well). Clearly it's possible - I just need to figure it out (it's not easy to debug).

MartyMacGyver, 2019-07-08:
^^^ Progress... FastLED is a somewhat different animal (e.g., I'm still not sure what the point of sequential "controllers" is) but it helped to see how they handled the semaphores and so on.

[image]

MartyMacGyver, 2019-07-14:
Closing as the new version of this library supports this now. Thank you again for taking the time to report and discuss this!

closed 2019-07-14 by MartyMacGyver

eyer1951, 2019-07-15:
Thanks Martin, excellent work!

Mark

MartyMacGyver, 2019-07-15:
It was worth getting it done, mainly because FastLED doesn't have RGBW yet (and as noted mine is a slimmer library, being specifically for the ESP32 family).

There is still work to be done as I clean up the API, etc. - and I haven't tested with esp-idf (but it's more a question of gluing it all together than functionality - there's nothing Arduino-specific in the core logic). But I'm glad it finally works the way I intended.