MartyMacGyver/ESP32-Digital-RGB-LED-Drivers #33: Can not run examples.

HudsonProdigy, 2019-08-01:
Hello, I can not get examples to run. It seems like part of the library is missing? where is function where is function digitalLeds_updatePixels(pStrand)? must we implement this on our own?

when attempting to comile example as given i get the following errors:

error: too few arguments to function 'digitalLeds_resetPixels'
error: implicit declaration of function 'digitalLeds_updatePixels'

i can fix the reset pixels issue by passing in an argument for num_strands, but i am unsure what is going on with update pixels?

any help would be appreciated thanks.

MartyMacGyver, 2019-08-01:
I need to update the readme - the IDF side is NOT working and I haven't had time to dig into why. I will eventually update the demos (likely with just one demo) assuming I can get it to work readily in the current IDF (it keeps changing and breaking things, while the Arduino side "just works").

HudsonProdigy, 2019-08-01:
That’s to bad. Was excited for this library because of idf compatible. Can not find a library that works to drive digital LEDs from esp32 idf. We’re those old functions removed from library?

MartyMacGyver, 2019-08-02:
Update: I spent about three hours tonight trying to get this to work in the ESP-IDF 3.2 line... so far, lots of arcane errors because why not make `min()` and `max()` impossible to redefine/macro like before? And the old build and make framework wasn't confusing enough so that got a major upgrade to a new level of obfuscation. And examples? Sure, but almost entirely in C because C++ is far too bleeding edge for that!

In short, I remember now EXACTLY why I quit trying to keep up with the IDF. v4.0 will be even more fun (assuming I get this to even work with 3.2) because it depends on Python... Python 3? Nope! Python 2.7, which has a hard EOL in just over 3 months.

I've got a lot of feature work I'd like to be doing to clean up and improve the driver and its attendant libraries and examples... As time permits I'll poke around at this, but shoehorning this to work with the IDF is not a priority right now.