MartyMacGyver/ESP32-Digital-RGB-LED-Drivers #34: Max Brightness Setting Ignored

henryjw, 2019-08-24:
Is the `brightnessLimit` field respected in the `strand_t` implementation? I'm trying with the code below, but it doesn't seem to make a difference in the brightness of the LEDs.

```c
strand_t strand = {
    .rmtChannel = 0,
    .gpioNum = 32,
    .ledType = LED_WS2812B_V2,
    .brightLimit = 10, // Doesn't seem to matter what the value is. Brightness doesn't change
    .numPixels = 13,
    .pixels = null,
    ._stateVars = null
};
```

MartyMacGyver, 2019-08-24:
It's used in the demo but it's not a hard limit in the driver itself. (I'm not sure it should be as it adds cycles... It's currently up to the user to reference that setting or to ignore it.)

henryjw, 2019-08-24:
Thanks for the quick response. From the demo, I see that the "brightness level" is basically faked by setting the color. Is that accurate?

```c
anim_max = pStrand->brightLimit - anim_step;
...

color1 = pixelFromRGBW(anim_max, 0, 0, 0);
```

MartyMacGyver, 2019-08-25:
Yeah, it was a _very_ basic form of it.

I'll keep this issue open though - I think I either need to remove that from the strands or implement it at a lower level at some point.