MartyMacGyver/ESP32-Digital-RGB-LED-Drivers #36: channels coupled together

RoelHermus, 2019-12-27:
Hi Marty, I programmed an ESP32 with your DEMO1.ino program and it gives some problems.
I've connected three ledstring to the ESP32, on GPIO 13,14 and 15.
When I send some data to the first led-strand it works perfect. Sending data to the second strand its also send the same data to the first led-strand Sending data to led-strand 3, also strand 2 and 1 are filled with the same data. For a good look I modify some timing and number of led strings, but nothing more.

Is this an issue only with the demo,  is this something in the program or do I something wrong?

MartyMacGyver, 2019-12-27:
This sounds unusual to me. Is there a level shifter in the circuit? Are any wires twisted together?

RoelHermus, 2019-12-28:
Hello Martin, forget my problem, there is nothing wrong. But it's really hard to understand how your driver works. Today I spent a few hours figuring it out.
Tomorrow a new day and I hope with more success, if not ...?

closed 2019-12-28 by RoelHermus