MartyMacGyver/ESP32-Digital-RGB-LED-Drivers #37: What is the benefit over FastLed lib ?

DeeJayMX, 2020-05-28:
Everything is in the title,

Your library looks promizing but I just wanted to know what is the benefit to use RMT over the FastLed style library which more Software controled by the CPU itself.

Thanks for your great job 😊

veitk, 2020-05-28:
imho fastLED is now also using RMT.

MartyMacGyver, 2020-05-28:
Look closer still and you'll see that the RMT code in FastLED is a fork of mine.

I took the proof of concept and expanded it into the driver I have here, and it helped FastLED get the same capability. One thing mine does easily that FastLED doesn't do easily (yet) is RGBW.

I also contribute to FastLED and hope to see that functionality there someday, at which point mine will be redundant. But that might be a while as it'll mean a pretty big overhaul of FastLED to do RGBW directly.

MartyMacGyver, 2020-05-28:
Oh, and RMT is faster and much more efficient than bit-banging for non-SPI devices like the WS2812. It's unique to the ESP32 as far as I know. Otherwise, efficiency for other non-ESP devices can be found by driving in parallel and/or having a high clock rate. Or offloading the work to a dedicated coprocessor.

DeeJayMX, 2020-05-29:
Thanks for that clarifying answer, You should put theses messages on the readme.md.

Thanks

NirViaje, 2020-11-25:
is it memory costly for RMT?

> Look closer still and you'll see that the RMT code in FastLED is a fork of mine.
>
> I took the proof of concept and expanded it into the driver I have here, and it helped FastLED get the same [...]

MartyMacGyver, 2020-11-25:
> is it memory costly for RMT?

The RMT only has so much buffer room, which means more interrupt overhead to keep it fed. And it's pretty much unique to the ESP32 - if you want to do things with other chipsets it's usually some kind of GPIO or I2S bit-banging magic.

I've often thought it'd be nice to have some sort of SPI or I2C device (probably FPGA to start with) that can act as a basic protocol translator to WS2812 or similar devices - then instead of playing games figuring out timing you just set some registers and pipe your data in, and it deals with clocking that data out. There would be some small latency but the benefit would be compatibility with virtually any microcontroller.

Too many projects, too little time...