MartyMacGyver/ESP32-Digital-RGB-LED-Drivers #38: (ESP-IDF) gRmtSem is never given when Heap Corruption Detection is set to Light Impact

benpeoples, 2020-07-31:
This is with ESP-IDF v4.0... 

With Heap Corruption Detection set to Comprehensive, everything works (reasonably) fine.   

With HCD at Basic, I get a LoadProhibited at line 297 of esp32_digital_led_lib.cpp (in digitalLeds_resetPixels())

With HCD in Light Impact, the system hangs at line 367 of esp32_digital_led_lib.cpp (in digitalLeds_drawPixels()), waiting to take gRmtSem.    

Any idea where to look to debug this?   

(Big picture, I'm getting some glitching and so I'm trying to figure out how to tune out the glitching...)

MartyMacGyver, 2020-07-31:
Since this codebase is for the most part contained within FastLED now, I haven't been actively maintaining it (and I haven't worked directly with the IDF directly in a long while now). I've not used HCD yet either, so I can't begin to speculate.

benpeoples, 2020-07-31:
All good, thanks!   I'll try porting over to FastLED and see if that solves my issues.    I think there was some particular reason I didn't want to use it, but I think maybe I've worked around that particular problem.

MartyMacGyver, 2020-08-01:
There has been a lot of development on the code since it was forked into there. That said, as time permits I may revisit this issue (I like the ESP32, but the Teensy 4.1 is also a platform I use). The one reason I keep this around is for RGBW (particularly on Arduino), which FastLED doesn't support yet.